Lora-net/LoRaMac-node #626: Unable to Receive LoRaWAN Messages using SX1276RF1JAS

KhaledAljehani, 2018-12-24:
Hi,

I connected SX1276RF1JAS to NUCLEO-L4, I am using ST source code for LoRaWAN 1.02, and 868 region.
For SX1276RF1JAS, I built it for sx1276mb1las target board.

My problem is I can send Confirmed/Unconfirmed LoRaWAN class A and C messages successfully, but unfortunately no reception at all.
The receive windows interrupts get executed all time except in TX (DIO0 -rxDone, DIO1 - rxTimeout). Also, I connected an antenna pin to both FEM_CTX and RXTX pins, but nothing changed.

I have suspects in the antenna pin, does it indicate no switch between RX and TX operations? Or caused by another source?

Looking forward to your support.

Best regards.

KhaledAljehani, 2018-12-26:
Any response?

mluis1, 2019-02-12:
Unfortunately we aren't aware of the modifications that ST made in his fork.

We would recommend that you ask your question directly to ST?

closed 2019-02-12 by mluis1