Lora-net/LoRaMac-node #627: LoRa Node can not receive or send any data after several hours

xiongjiangj, 2018-12-25:
Hi, I met a problem that after several hours test (I made the LoRa Node keep sending and receiving data with the Server, every 5 seconds send a confirmed data to the Server, and receive ACK data), the LoRa Node will never send or receive any data from the Server.
I found the root cause of this problem is that the state of SX1276 can not match with the TxTimeoutTimer and RxTimeoutTimer, when I send a data, the node will get a rxtimeout, so I fix this problem as follows,
```
void SX1276SetRx( uint32_t timeout )
{
    bool rxContinuous = false;
    TimerStop( &TxTimeoutTimer );     // new line
    ......
}
```

```
void SX1276SetTx( uint32_t timeout )
{
    TimerStop( &RxTimeoutTimer );      // new line
    TimerSetValue( &TxTimeoutTimer, timeout );
    ......
}
```

I use the code version is 4.4.2 - rc1,

mluis1, 2019-02-12:
We aren't able to reproduce this issue. We have run a test in the same conditions and after 2 days it was still working as expected.

In order to be on the safe side we think that your proposal can be added to the radio drivers.

trick2011, 2019-02-26:
Could you add the relevant patch that is to be committed?

closed 2019-03-05 by mluis1