Lora-net/LoRaMac-node #628: The Timer is not valid, but used by the LoRaMac

xiongjiangj, 2018-12-25:
Hi, I met a problem that when LoRa Node lost connect and rejoin the network, the node received a join accept message, but not processed by the LoRaMacProcess( ) function, because MacCtx.MacFlags.Bits.MacDone is not set to 1 in ProcessRadioRxDone( ) function, bacause MacCtx.AckTimeoutTimer.IsRunning is still true. But I have stop MacCtx.AckTimeoutTimer before the node send the join request, so MacCtx.AckTimeoutTimer has been removed from the TimerList, but still used in ProcessRadioRxDone( ), this is the root cause of this problem, and I fix this problem as flows,

```
void TimerStop( TimerEvent_t *obj )
{
    ......
    // List is empty or the obj to stop does not exist
    if( ( TimerListHead == NULL ) || ( obj == NULL ) )
    {
        // new lines 
        if (obj != NULL) {
            obj->IsRunning = false;
        }

        CRITICAL_SECTION_END( );
        return;
    }
    ......
    else // Stop an object within the list
    {
        obj->IsRunning = false;     // new line
        while( cur != NULL )
        {
            if( cur == obj )
        ......
```

mluis1, 2019-02-12:
Which branch/version are you using?

In develop branch we have updated the timer.c/h module. Could you please try the code on develop branch and verify that your issue is solved.

The new code is as shown below. As you can see the `IsRunning` variable doesn't exist anymore. It has been replace by `IsStarted` and by `IsNext2Expire`.

```c
void TimerStop( TimerEvent_t *obj )
{
    CRITICAL_SECTION_BEGIN( );

    TimerEvent_t* prev = TimerListHead;
    TimerEvent_t* cur = TimerListHead;

    // List is empty or the obj to stop does not exist
    if( ( TimerListHead == NULL ) || ( obj == NULL ) )
    {
        CRITICAL_SECTION_END( );
        return;
    }

    obj->IsStarted = false;

    if( TimerListHead == obj ) // Stop the Head
    {
        if( TimerListHead->IsNext2Expire == true ) // The head is already running
        {
            TimerListHead->IsNext2Expire = false;
            if( TimerListHead->Next != NULL )
            {
                TimerListHead = TimerListHead->Next;
                TimerSetTimeout( TimerListHead );
            }
            else
            {
                RtcStopAlarm( );
                TimerListHead = NULL;
            }
        }
        else // Stop the head before it is started
        {
            if( TimerListHead->Next != NULL )
            {
                TimerListHead = TimerListHead->Next;
            }
            else
            {
                TimerListHead = NULL;
            }
        }
    }
    else // Stop an object within the list
    {
        while( cur != NULL )
        {
            if( cur == obj )
            {
                if( cur->Next != NULL )
                {
                    cur = cur->Next;
                    prev->Next = cur;
                }
                else
                {
                    cur = NULL;
                    prev->Next = cur;
                }
                break;
            }
            else
            {
                prev = cur;
                cur = cur->Next;
            }
        }
    }
    CRITICAL_SECTION_END( );
}
```

xiongjiangj, 2019-02-15:
I used the version of 4.4.2, so I will upgrade my code ASAP, Thanks!

closed 2019-03-22 by mluis1