Lora-net/LoRaMac-node #631: Stack nrF52

Sani300, 2019-01-03:
Hi,

Is that possible to have stack Lora compatible to nrF52 from Nordic Semiconductor? Are you developping other target ?

Sani300

mluis1, 2019-01-04:
A port to NRF52 platform is possible.

As of today we have no intention to add new platforms. 
Adding a new platform means a lot of work for us. 

The issue is not the porting itself which should be quite straight forward. The issue is more on the maintenance work afterwards.

When a release is performed we need to check that everything builds correctly as well as we need to run one example on each supported platform.

Another issue is also that currently we don't have a nrf52 platform available.

nestorayuso, 2019-01-04:
here is a quick list of ports for nRF52:

https://github.com/gluedig/nrf52-lora
https://github.com/RAKWireless/RAK813_LoRaNode
https://github.com/RAKWireless/RAK813-BreakBoard

Sani300, 2019-01-06:
Ok mluis1, I can understand you don’t add a new platform!

Yes, I have already see those project. The thing is that we would like to use sx1261. And we can’t find any project with this component. 

So we have change those project. But each time we send a message by SPI to sx1261, we have 0xB2. I can’t find any information about that in datasheet. Do you know why?!

x893, 2019-01-07:
https://github.com/x893/LoRaMac-node/tree/feature/class_b/Keil/RAK8211/LoRaMac/classB