Lora-net/LoRaMac-node #632: Problems Adding I2C Sensor to STM32L073 Module

matomback, 2019-01-03:
Hello,

I am having some difficulty adding an I2C sensor (SHT21) to the STM32L073 module using the VSCode IDE and the ClassA application. I do not have a ton of experience with the Loramac_node stack yet, but I was able to get the ClassA application working and sending uplinks to our Gateway. Any help is greatly appreciated!

I used the NAMote72 ClassA application for reference, as it implements a temperature sensor (MPL3115) over I2C.

I first created sht21.c and sht21.h in the peripherals directory. They implement some simple functions that call functions in i2c.c which subsequently calls the HAL functions in i2c-board.c. This is the same basic structure as in the NAMote72.

The issue occurs when I try to add the 'I2cInit' function to the board.c file (under the BoardInitMcu subroutine, in the same location that it is used in the NAMote72). I get an 'undefined reference' to both I2cMcuInit and I2cMcuFormat, though I did not change either of these functions. Build output below:

> [build] Starting build
[proc] Executing command: "C:/Program Files/CMake/bin/cmake.exe" --build c:/GIT/LoRa/LoRaMac-node_fork/LoRaMac-node/build --config Release --target all -- -j 6
[build] [  6%] Building C object src/radio/CMakeFiles/radio.dir/sx126x/radio.c.obj
[build] [  3%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mag3110.c.obj
[build] [  3%] Building C object src/peripherals/CMakeFiles/peripherals.dir/gpio-ioe.c.obj
[build] [  6%] Building C object src/system/CMakeFiles/system.dir/adc.c.obj
[build] [  6%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMac.c.obj
[build] Scanning dependencies of target NucleoL073
[build] [  7%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/board.c.obj
[build] [  8%] Building C object src/system/CMakeFiles/system.dir/delay.c.obj
[build] [ 11%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mma8451.c.obj
[build] [ 10%] Building C object src/radio/CMakeFiles/radio.dir/sx126x/sx126x.c.obj
[build] [ 12%] Building C object src/system/CMakeFiles/system.dir/eeprom.c.obj
[build] [ 13%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacAdr.c.obj
[build] [ 15%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mpl3115.c.obj
[build] [ 16%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/delay-board.c.obj
[build] [ 17%] Building C object src/system/CMakeFiles/system.dir/fifo.c.obj
[build] [ 17%] Built target radio
[build] [ 18%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/eeprom-board.c.obj
[build] [ 20%] Building C object src/peripherals/CMakeFiles/peripherals.dir/pam7q.c.obj
[build] [ 22%] Building C object src/system/CMakeFiles/system.dir/gpio.c.obj
[build] [ 22%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacClassB.c.obj
[build] [ 23%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/gpio-board.c.obj
[build] [ 25%] Building C object src/peripherals/CMakeFiles/peripherals.dir/sht21.c.obj
[build] [ 26%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacCommands.c.obj
[build] [ 27%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/lpm-board.c.obj
[build] [ 28%] Building C object src/system/CMakeFiles/system.dir/gps.c.obj
[build] [ 30%] Building C object src/peripherals/CMakeFiles/peripherals.dir/soft-se/aes.c.obj
[build] [ 31%] Building C object src/system/CMakeFiles/system.dir/i2c.c.obj
[build] [ 32%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacConfirmQueue.c.obj
[build] [ 33%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/rtc-board.c.obj
[build] [ 35%] Building C object src/peripherals/CMakeFiles/peripherals.dir/soft-se/cmac.c.obj
[build] [ 36%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/spi-board.c.obj
[build] [ 37%] Building C object src/system/CMakeFiles/system.dir/nvmm.c.obj
[build] [ 38%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacCrypto.c.obj
[build] [ 40%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/uart-board.c.obj
[build] [ 41%] Building C object src/system/CMakeFiles/system.dir/systime.c.obj
[build] [ 42%] Building C object src/peripherals/CMakeFiles/peripherals.dir/soft-se/soft-se.c.obj
[build] [ 43%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacFCntHandler.c.obj
[build] [ 45%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/sysIrqHandlers.c.obj
[build] [ 46%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacParser.c.obj
[build] [ 47%] Building C object src/system/CMakeFiles/system.dir/timer.c.obj
[build] [ 48%] Building C object src/peripherals/CMakeFiles/peripherals.dir/sx1509.c.obj
[build] [ 50%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/utilities.c.obj
[build] [ 51%] Building C object src/system/CMakeFiles/system.dir/uart.c.obj
[build] [ 52%] Building C object src/peripherals/CMakeFiles/peripherals.dir/sx9500.c.obj
[build] [ 53%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacSerializer.c.obj
[build] [ 56%] Building C object src/mac/CMakeFiles/mac.dir/region/Region.c.obj
[build] [ 55%] Building ASM object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/cmsis/arm-gcc/startup_stm32l073xx.s.obj
[build] [ 57%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionAS923.c.obj
[build] [ 57%] Built target system
[build] [ 58%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/cmsis/system_stm32l0xx.c.obj
[build] [ 60%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionAU915.c.obj
[build] [ 61%] Built target peripherals
[build] [ 61%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionCN470.c.obj
[build] [ 62%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal.c.obj
[build] [ 63%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_adc.c.obj
[build] [ 66%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_adc_ex.c.obj
[build] [ 66%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionCN779.c.obj
[build] [ 68%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_cortex.c.obj
[... 198 lines cut ...]
[build] mingw32-make.exe[1]: *** Waiting for unfinished jobs....
[build] [ 40%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacParser.c.obj
[build] [ 41%] Building C object src/system/CMakeFiles/system.dir/timer.c.obj
[build] [ 43%] Building C object src/peripherals/CMakeFiles/peripherals.dir/soft-se/soft-se.c.obj
[build] [ 44%] Building C object src/peripherals/CMakeFiles/peripherals.dir/sx1509.c.obj
[build] [ 45%] Building C object src/system/CMakeFiles/system.dir/uart.c.obj
[build] [ 46%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacSerializer.c.obj
[build] [ 49%] Building C object src/mac/CMakeFiles/mac.dir/region/Region.c.obj
[build] [ 49%] Building C object src/peripherals/CMakeFiles/peripherals.dir/sx9500.c.obj
[build] [ 50%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionAS923.c.obj
[build] [ 51%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionAU915.c.obj
[build] [ 53%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionCN470.c.obj
[build] [ 53%] Built target system
[build] [ 54%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionCN779.c.obj
[build] [ 55%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionCommon.c.obj
[build] [ 55%] Built target peripherals
[build] [ 56%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionEU433.c.obj
[build] [ 58%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionIN865.c.obj
[build] [ 59%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionEU868.c.obj
[build] [ 61%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionRU864.c.obj
[build] [ 61%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionKR920.c.obj
[build] [ 62%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionUS915.c.obj
[build] [ 62%] Built target mac
[build] mingw32-make.exe: *** [Makefile:83: all] Error 2
[build] Build finished with exit code 2 <

mluis1, 2019-01-04:
Could you please uncomment line 77 on file `src/boards/NucleoL053/cmsis/stm32l0xx_hal_conf.h` file.

By doing this the I2C module gets enabled for the HAL.
After this change the project should build correctly.

Before:
```c
#define HAL_MODULE_ENABLED  
//#define HAL_ADC_MODULE_ENABLED   
//#define HAL_COMP_MODULE_ENABLED   
//#define HAL_CRC_MODULE_ENABLED   
//#define HAL_CRYP_MODULE_ENABLED   
//#define HAL_DAC_MODULE_ENABLED   
//#define HAL_FIREWALL_MODULE_ENABLED   
//#define HAL_I2S_MODULE_ENABLED   
//#define HAL_IWDG_MODULE_ENABLED   
//#define HAL_LCD_MODULE_ENABLED   
//#define HAL_LPTIM_MODULE_ENABLED   
//#define HAL_RNG_MODULE_ENABLED   
#define HAL_RTC_MODULE_ENABLED   
#define HAL_SPI_MODULE_ENABLED   
//#define HAL_TIM_MODULE_ENABLED   
//#define HAL_TSC_MODULE_ENABLED   
#define HAL_UART_MODULE_ENABLED   
//#define HAL_USART_MODULE_ENABLED   
//#define HAL_IRDA_MODULE_ENABLED   
//#define HAL_SMARTCARD_MODULE_ENABLED   
//#define HAL_SMBUS_MODULE_ENABLED   
#define HAL_WWDG_MODULE_ENABLED   
//#define HAL_PCD_MODULE_ENABLED   
#define HAL_GPIO_MODULE_ENABLED
#define HAL_DMA_MODULE_ENABLED
//#define HAL_I2C_MODULE_ENABLED           // <=== To be uncommented
#define HAL_RCC_MODULE_ENABLED
#define HAL_FLASH_MODULE_ENABLED
#define HAL_PWR_MODULE_ENABLED
#define HAL_CORTEX_MODULE_ENABLED
```

After:
```c
#define HAL_MODULE_ENABLED  
//#define HAL_ADC_MODULE_ENABLED   
//#define HAL_COMP_MODULE_ENABLED   
//#define HAL_CRC_MODULE_ENABLED   
//#define HAL_CRYP_MODULE_ENABLED   
//#define HAL_DAC_MODULE_ENABLED   
//#define HAL_FIREWALL_MODULE_ENABLED   
//#define HAL_I2S_MODULE_ENABLED   
//#define HAL_IWDG_MODULE_ENABLED   
//#define HAL_LCD_MODULE_ENABLED   
//#define HAL_LPTIM_MODULE_ENABLED   
//#define HAL_RNG_MODULE_ENABLED   
#define HAL_RTC_MODULE_ENABLED   
#define HAL_SPI_MODULE_ENABLED   
//#define HAL_TIM_MODULE_ENABLED   
//#define HAL_TSC_MODULE_ENABLED   
#define HAL_UART_MODULE_ENABLED   
//#define HAL_USART_MODULE_ENABLED   
//#define HAL_IRDA_MODULE_ENABLED   
//#define HAL_SMARTCARD_MODULE_ENABLED   
//#define HAL_SMBUS_MODULE_ENABLED   
#define HAL_WWDG_MODULE_ENABLED   
//#define HAL_PCD_MODULE_ENABLED   
#define HAL_GPIO_MODULE_ENABLED
#define HAL_DMA_MODULE_ENABLED
#define HAL_I2C_MODULE_ENABLED           // <=== Uncommented
#define HAL_RCC_MODULE_ENABLED
#define HAL_FLASH_MODULE_ENABLED
#define HAL_PWR_MODULE_ENABLED
#define HAL_CORTEX_MODULE_ENABLED
```

matomback, 2019-01-09:
That did it, the sensor is working now, thanks very much. I was having trouble following the error output but figured it was something small that I looked over. Appreciate the help!

closed 2019-01-09 by matomback