Lora-net/LoRaMac-node #635: ChannelsMaskRemaining is reset for every join retry

lucaszfolle, 2019-01-07:
When calling the LoRaMacMlmeRequest function to execute a join procedure, it calls ResetMacParameters, which also resets the ChannelsMaskRemaining array. This makes it impossible to perform the recommended channel sequence for join retries (US915 and AU915; specs 1.0.3).
Or is there another way of retransmitting join requests without calling LoRaMacMlmeRequest? Since it doesn't handle retries for MLME messages as it does for MCPS messages.

djaeckle, 2019-02-07:
Hi lucaszfolle,

which version are you using? In the function `ResetMacParameters`, the MAC resets the channels with `INIT_TYPE_RESTORE_DEFAULT_CHANNELS`. This should not lead to a reset ot the `ChannelsMaskRemaining`. Or do i miss something?

lucaszfolle, 2019-02-08:
Hi djaeckle,

the version I was using was the 4.4.2-rc.1, and indeed the current version makes a copy-and of the default channels into the ChannelsMaskRemaining. On the 4.4.2-rc.1 it was just copying as is.
Thanks!

closed 2019-02-08 by lucaszfolle