Lora-net/LoRaMac-node #661: Question Regarding - system/timer

axu217, 2019-01-15:
Is I do not need to operate in sleep mode, can timer be implemented using a hardware timer instead of the RTC?

If RTC time and date is changed, will timer alarms from system/timer that already exist be affected?

mluis1, 2019-01-16:
Yes, you may use whatever timer you want.

The STM32 RTC handling is quite complex because we need to convert a 64 bits variable containing the counter value to BCD and from BCD to a 64 bits value.

But, the only requirement is to have an HW counter (clocked by a precise source XTAL) that has the possibility to generate interrupts when compared to a specific value.

In order to use your HW timer you just need to implement the `rtc-board.h` API. 
In the new `rtc-board.c` file you must setup your timer in such way that it fulfills the API contract.

The SAML21 platform uses an RTC which is similar to a normal timer. I think that you can get your inspiration from it.

axu217, 2019-01-16:
Thank you for the quick response! If I use STM's HAL library to set the RTC's date/time, will I have to reset the system timers from LoRa-Mac?

mluis1, 2019-02-12:
If you use a different HW timer as base for the `time.c/h` then you can do whatever you would like with the RTC hardware.

In case you want to keep using the RTC as it is today then you need to use the `time.c/h` API in order to set the current time and date.

closed 2019-03-22 by mluis1