Lora-net/LoRaMac-node #662: reduce firmware size

wireless-road, 2019-01-18:
Hi!
There are no any issues or bugs. Firmware works like a charm.
But after adding user logic to source code of loramac-node  I got only 280 byte of free memory:
[image]
For 64kB flash size device.
So I am wondering for any solution of reducing firmware size. I already set Optimization level -Os (for size).
The only thing I have is replacing HAL with LL.
Maybe there are some more quick ways of solving this task?)

temporaryaccount, 2019-01-22:
> Maybe there are some more quick ways of solving this task?)

Change the microcontroller model so it has at least 128 KB of flash.

Really, porting to LL or CMSIS might gain you a couple of KBs, but there is nothing much that could be done after that. You could try to turn on LTO (link-time-optimizations) to squeeze some kilobytes (maybe..), but again, not a real solution.

nestorayuso, 2019-02-12:
as asked here, https://github.com/Lora-net/LoRaMac-node/issues/595 it would be fine to be able to disable ClassC and Multicast at compile time using 
#ifdef LORAMAC_CLASSC_ENABLED ... or #ifdef LORAMAC_MULTICAST_ENABLED

mluis1, 2019-04-02:
No activity we close the issue.

closed 2019-04-02 by mluis1