Lora-net/LoRaMac-node #667: Question about class C: management of Uplinks and Downlinks on RX2

Adamzat, 2019-01-28:
Hi every one, 

I'm using the Lora Stack version 4.4.0 and I have a question about the operation of class C.
I would like to know how the Lora Mac manages at the same time the reception of downlinks (on the continuous Rx2) and sending a new Uplink request needed by the application ? What API is used? 
If my application needs to send a new uplink using the "LoRaMacMcpsRequest", is there a risk that timers are running (like RxWindowTimer2 or AckTimeoutTimer ) and I have to stop them ?
How it is managed automatically ? uplinks can be lost when DL is received ?

Please help.

Thank you

djaeckle, 2019-02-07:
Hi Adamzat,

thanks for your question. The MAC opens the reception windows automatically. Also, it handles the timeouts automatically. Thus, if your device operates in Class C, you could perform a transmission by calling LoRaMacMcpsRequest at any time (when the MAC is idle). This means a transmission has higher priority than listening to the continuous RX window.

Adamzat, 2019-02-07:
Hi djaeckle,
Thank you for your quick reply. 
I have observed during tests that when I stressed my module to send data periodically in class C (each 5 seconds or less). Sometimes I fall in the state LORAMAC_STATUS_BUSY (at this state no downlink were received on RX2), so I don't understand why the LoraMac is busy !

I did not have this problem in class A.

Any Idea ?

mluis1, 2019-03-22:
The MAC layer enforces the uplinks to be sent at least every 5 seconds. If you try to use a smaller periodicity then the MAC layer will return LORAMAC_STATUS_BUSY because it is still handling the previous uplink.
Please note that this is true if no downlink happened in one of the 2 ClassA reception windows

Adamzat, 2019-03-25:
Hi mluis, 
Thank you for your reply. Yes, the problem happens when I tried to send data in a period less than 5 seconds.
I close the issue.

closed 2019-03-25 by Adamzat