Lora-net/LoRaMac-node #668: Class C window not opened after receiving a frame not destined for the end device

snelson-senet, 2019-01-29:
**Configuration**
- Board: NAMote72
- Application: LoRaMac/classC (develop branch, commit 57ecd2 (01/14/2019)
- Downlinks to the device at a 20 second interval 

**Issue** 
I frequently observe the end device go from receiving downlinks to not receiving any until after transmitting an uplink.

I've determined this failure occurs after receiving a frame in the class C Rx window that is not for the device.  What looks different in this event from other Rx abort conditions is that PrepareRxDoneAbort() is not called.

```
macCryptoStatus = LoRaMacCryptoUnsecureMessage( addrID, address, fCntID, downLinkCounter, &macMsgData );
if( macCryptoStatus != LORAMAC_CRYPTO_SUCCESS )
{
    if( macCryptoStatus == LORAMAC_CRYPTO_FAIL_ADDRESS )
    {
        // We are not the destination of this frame.
        MacCtx.McpsIndication.Status = LORAMAC_EVENT_INFO_STATUS_ADDRESS_FAIL;

       // Abort the reception, if we are not in RX_SLOT_WIN_CLASS_C
       if( MacCtx.McpsIndication.RxSlot != RX_SLOT_WIN_CLASS_C )
       {
           PrepareRxDoneAbort( );
       }
   }
   else
   {
       // MIC calculation fail
       MacCtx.McpsIndication.Status = LORAMAC_EVENT_INFO_STATUS_MIC_FAIL;
       PrepareRxDoneAbort( );
   }
```
I changed this code  to call PrepareRxDoneAbort()  regardless of RxSlot type and now see the Class C window re-opened and downlinks are received.

djaeckle, 2019-02-07:
Hi snelson-senet,

thanks for the report. We will double check your proposal.

mluis1, 2019-03-22:
Fix applied with commit b32b2177381f73e8a57963a5b42640922f9a0a84

closed 2019-03-22 by mluis1