Lora-net/LoRaMac-node #669: Need help with compiling this project

Gep93, 2019-01-30:
Hey,

I am sorry for asking such a noob question and for this not being and actual issue but I am totally in the dark about how to proceed after downloading the code from this github.

I would like to test class A device code with my loraserver network but have no idea how to compile this code and get it to my node (murata type ABZ module).

I have a code from ST which I have gotten from someone and I upload that code to my node with KEIL. The code was already set to go so I didn't have to create the project and do everything myself. Since STs code has some weird behaviour I would like to test the code from here but like I said I have no idea how to do it.

Could somone be so kind and walk me through it?

Thanks,
G

Gep93, 2019-02-04:
Ok, so now I have Cmake and MinGW on my computer (Windows).

I used this command in cmd:

`cmake -DCMAKE_TOOLCHAIN_FILE="cmake/toolchain-arm-none-eabi.cmake" -DTOOLCHAIN_PREFIX="/path/to/the/toolchain" -G "MinGW Makefiles"`

I left first path as is, and changed the second path to the path of "cmake" folder located in ...\LoRaMac-node-4.4.1\cmake

Am I doing this right?

djaeckle, 2019-02-07:
Hi Gep93,

you need to setup DTOOLCHAIN_PREFIX to your local path of your toolchain. E.g. if you are on windows and you have downloaded '6_2017-q2-update' and you have installed it directly in 'C:\', you should apply  the path 'TOOLCHAIN_PREFIX":"c:/6_2017-q2-update/".

djaeckle, 2019-03-28:
Hi Gep93,

have you solved it? Can we close this issue?

Gep93, 2019-03-28:
Hey, 

sorry I haven't. Had to work on some other stuff regarding LoRa and will get back to this issue soon.

If this thread could stay open for now I would greatly appreaciate it!

mluis1, 2019-06-07:
As there is no updates we decided to close the issue.

closed 2019-06-07 by mluis1