Lora-net/LoRaMac-node #670: Join ABP after Successfully join as OTA ?

labviewegypt, 2019-01-31:
I am using LORAWAN and after I joined as OTA, how to join as ABP using the generated season keys which generated after joined OTA?
I read before thatits better first to join as OTA then join as ABP using the generated keys.
Also, how to change the TX data rate after joined the network?

nestorayuso, 2019-02-01:
if you follow the loramac main.c example,
https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/LoRaMac/classA/B-L072Z-LRWAN1/main.c#L970
the context is stored in non-volatile-memory and one of the first things done in the state machine is to try to restore the previous state of the lorawan session.

After a reset basically you need to do:
LoRaMacInitialization( &macPrimitives, &macCallbacks, ACTIVE_REGION );
NvmCtxMgmtRestore( );
LoRaMacStart( );
and you are done, no need to specifically perform a ABP join.

before power off your device you must store the state in non-volatile-memory using NvmCtxMgmtStore( )

labviewegypt, 2019-02-01:
> if you follow the loramac main.c example,
> https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/LoRaMac/classA/B-L072Z-LRWAN1/main.c#L970
> the context is stored in non-volatile-memory and [...]

Hi
Thank you for your reply
The stack version i have not have these api "NvmCtxMgmtRestore( )"
i am using st lorwan stack version 1.2.1

mluis1, 2019-02-08:
As you look to be using the ST fork of this stack we would recommend that you directly contact ST for support. 
Or you can also start using the version provided here.

closed 2019-02-08 by mluis1