Lora-net/LoRaMac-node #671: AS923

coljander, 2019-02-04:
Hi

I attended Miguel's Q&A at the recent TTI conference in Amsterdam.  We brought up some issues with the AS923 region and he said that this would all be handled in the 1.0.4 branch.  I've had a look and there doesn't appear to be a 1.0.4 branch.   Has this already been merged into develop or 5.0.0?  Or did I misunderstand and it is version 1.0.4 of a spec?
Can you please supply a commit or link that explains what 1.0.4 changes?

Thanks

Colin

mluis1, 2019-02-04:
The LoRaWAN specification 1.0.4 is currently being discussed by the Alliance technical committee.
We will create a branch and work on it once the specification is released.

BenAtPip, 2019-02-10:
> Hi
>
> I attended Miguel's Q&A at the recent TTI conference in Amsterdam. We brought up some issues with the AS923 region and he said that this would all be handled in the 1.0.4 branch. I've had a look [...]

Would you have any details on the issues you have identified for AS923 available Colin?

Ben

closed 2019-04-02 by mluis1