Lora-net/LoRaMac-node #673: Visual studio Build and debug Error

sagar-patel-sls, 2019-02-06:
Hii @ConnyBusy 
I also got build error in VS 2017 Professional.

Please see attached screenshot
[image]

Thanks

ConnyBusy, 2019-02-06:
Hi @sagar-patel-sls,
it is highly recommended to provide as much information as possible when you need help. Generally, screenshots like yours above aren't very useful. Did you go through the steps in the documentation on how to build?

sagar-patel-sls, 2019-02-07:
> Hi @sagar-patel-sls,
> it is highly recommended to provide as much information as possible when you need help. Generally, screenshots like yours above aren't very useful. Did you go through the steps in the documentation on how to build?

Hii @ConnyBusy Thank you for your reply
Yes, I go through the steps in the documentation  #555 ([Cmake Error](https://github.com/Lora-net/LoRaMac-node/issues/555))

Please find attached CMakeCache & CMakeList for more information
[CMakeLists.txt](https://github.com/Lora-net/LoRaMac-node/files/2839294/CMakeLists.txt)
[CMakeCache.txt](https://github.com/Lora-net/LoRaMac-node/files/2839296/CMakeCache.txt)

Thank You

ConnyBusy, 2019-02-07:
Hi @sagar-patel-sls,
there seems to be no errors in your CMakeCache.txt. What command are you using to build the binaries?
It's advisable to use the command line. i.e. After running CMake command just run make. You could also share a screenshot of your project directory after you run CMake

sagar-patel-sls, 2019-02-07:
Hii @ConnyBusy 
i am build using Visual studio build menu
[image]

VS Solution Explorer 
[image]

Project Directory
[image]

[image]

Thank you

ConnyBusy, 2019-02-07:
the target seems to have been built correctly. The LoRaMac-classA.bin is all you need to upload to you Node.

sagar-patel-sls, 2019-02-07:
> the target seems to have been built correctly. The LoRaMac-classA.bin is all you need to upload to you Node.

Hii @ConnyBusy 
Thanks for your reply

As per your conversion
`I need to embed **LoRaMac-classA.bin** in my Node (Board)?`

ConnyBusy, 2019-02-07:
here is the documentation

[https://github.com/Lora-net/LoRaMac-node/blob/develop/Doc/development-environment.md](url)

sagar-patel-sls, 2019-02-07:
> here is the documentation
> 
> [https://github.com/Lora-net/LoRaMac-node/blob/develop/Doc/development-environment.md](url)
@ConnyBusy 
1. I need to embed LoRaMac-classA.bin in my Node (Board)?

ConnyBusy, 2019-02-07:
Yes. That's all you need. Just copy paste it to your Board which emulates a drive when you connect it to your PC via USB
[image]

sagar-patel-sls, 2019-02-14:
Hii, @ConnyBusy 

i am try to compile & build sourcecode to **VSCODE** and i got success in VSCODE but when i debug code  so it will throw below screenshot error.

[image]
==================
i have cloned  **feature/5.0.0 branch** for implementation of LoRaWAN 1.1 

can you please help me.

Thank You

mluis1, 2019-02-14:
The "Exception has occurred" message almost always happen.

You can just ignore it and and click on "Continue execution", "Step into", "Step over".

sagar-patel-sls, 2019-02-15:
Hii, @ConnyBusy @mluis1 

I got Below error while debugging code
```
###### ===== MLME-Confirm ==== ######
STATUS      : Tx timeout
```
[image]

Please help me..
Thanks

sagar-patel-sls, 2019-02-25:
Hii @ConnyBusy 

Anyone can help me?

Thanks

ConnyBusy, 2019-02-27:
Looks like your radio shield and board aren't matching. You must set them correctly in the CMakeLists.txt files

sagar-patel-sls, 2019-02-27:
Thank you @ConnyBusy 
Below is my configuration file of VSCODE

        // In case your GNU ARM-Toolchain is not installed under the default
        // path:
        //     Windows : No default path. Specify the path where the
        //               toolchain is installed. i.e:
        //               "C:/PROGRA~2/GNUTOO~1/72017-~1".
        //     Linux   : /usr
        //     OSX     : /usr/local
        // It is required to uncomment and to fill the following line.
        "TOOLCHAIN_PREFIX":"C:/Program Files (x86)/GNU Tools Arm Embedded/7 2018-q2-update",

        // In case your OpenOCD is not installed under the default path:
        //     Windows : C:/openocd/bin/openocd.exe
        //     Linux   : /usr/bin/openocd
        //     OSX     : /usr/local/bin/openocd
        // Please uncomment the following line and fill it accordingly.
        //"OPENOCD_BIN":"C:/openocd/bin/openocd.exe",

        // Specifies the path to the CMAKE toolchain file.
        "CMAKE_TOOLCHAIN_FILE":"E:/PROJECTS/LORA/Eclipse/LoRaMac-node-feature-5.0.0/cmake/toolchain-arm-none-eabi.cmake",

        // Determines the application. You can choose between:
        // LoRaMac (Default), ping-pong, rx-sensi, tx-cw and BootLoader.
        "APPLICATION":"LoRaMac",

        // Select LoRaMac sub project. You can choose between:
        // classA, classB or classC.
        "CLASS":"classA",

        // Switch for Class B support of LoRaMac:
        "CLASSB_ENABLED":"OFF",

        // Select the active region for which the stack will be initialized.
        // You can choose between:
        // LORAMAC_REGION_EU868, LORAMAC_REGION_US915, ..
        "ACTIVE_REGION":"LORAMAC_REGION_EU433",

        // Select the type of modulation, applicable to the ping-pong or
        // rx-sensi applications. You can choose between:
        // LORA or FSK
        "MODULATION":"LORA",

        // Target board, the following boards are supported:
        // NAMote72, NucleoL073 (Default), NucleoL152, NucleoL476, SAML21, SKiM880B, SKiM980A, SKiM881AXL and B-L072Z-LRWAN1.
        "BOARD":"NucleoL152",

        // MBED Radio shield selection. (Applies only to Nucleo platforms)
        // The following shields are supported:
        // SX1272MB2DAS, SX1276MB1LAS, SX1276MB1MAS, SX1261MBXBAS(Default), SX1262MBXCAS, SX1262MBXDAS.
        "MBED_RADIO_SHIELD":"SX1276MB1MAS",

        // Region support activation, Select the ones you want to support.
        // By default only REGION_EU868 support is enabled.
        "REGION_EU868":"OFF",
        "REGION_US915":"OFF",
        "REGION_CN779":"OFF",
        "REGION_EU433":"ON",
        "REGION_AU915":"OFF",
        "REGION_AS923":"OFF",
        "REGION_CN470":"OFF",
        "REGION_KR920":"OFF",
        "REGION_IN865":"OFF",
        "REGION_RU864":"OFF"

Thank you

ConnyBusy, 2019-02-27:
This how your CMakeLists.txt should look like.
[CMakeLists.txt](https://github.com/Lora-net/LoRaMac-node/files/2911530/CMakeLists.txt)

sagar-patel-sls, 2019-02-28:
Hii, @ConnyBusy 

i also added your changes to **CMakeLists.txt**.

but still i got below error
```
###### ===== MLME-Confirm ==== ######
STATUS      : Tx timeout
```

Can you please help me..!
Thank you

sagar-patel-sls, 2019-03-05:
Hii, @ConnyBusy @mluis1 

Can you please help me.

Thanks

ConnyBusy, 2019-03-05:
@sagar-patel-sls,
I am not sure about the changes you have made to the source on your side. It should work normally as you can see below:

###### ===== MCPS-Request ==== ######
STATUS      : OK

###### ===== MCPS-Confirm ==== ######
STATUS      : OK

###### ===== UPLINK FRAME 1 ==== ######

CLASS       : A

TX PORT     : 2
TX DATA     : UNCONFIRMED
00

DATA RATE   : DR_0
U/L FREQ    : 433375000
TX POWER    : 0
CHANNEL MASK: 0007

###### ===== MCPS-Request ==== ######
STATUS      : OK

sagar-patel-sls, 2019-03-05:
@ConnyBusy 
please find attached my Source Code of feature **`feature/5.0.0 branch`**
[LoRaMac-node-feature-5.0.0.zip](https://github.com/Lora-net/LoRaMac-node/files/2929956/LoRaMac-node-feature-5.0.0.zip)

i am working on LoRaWAN V1.1 so i download source code of **`feature/5.0.0 branch`**

Please help me...

ConnyBusy, 2019-03-05:
@sagar-patel-sls ,

I am using the source code you have here. 

why on earth did you do this?? The remarks says they are generated automatically.

[image]

Leave it this way:

[image]

Your LoRaMAC ACTIVE_REGION is set to EU868 instead of EU433:

[image]

sagar-patel-sls, 2019-03-05:
@ConnyBusy 

Thank you for your replay i will change my code as per your message and try debugging.

Thank you

sagar-patel-sls, 2019-03-05:
Hii @ConnyBusy 

Sorry for disturbing you.

but i still got same error **`TX TIMEOUT`**

See below screenshot

[image]

Sorry..

ConnyBusy, 2019-03-05:
[image]

sagar-patel-sls, 2019-03-05:
@ConnyBusy 

**`SLS_TEMPERATURE_APPLICATION`** is temperature calculation source code.
we have developed source code for our temperature sensor

sagar-patel-sls, 2019-03-09:
Hii @ConnyBusy 

1. **For V1.1 you have used PA_BOOST?**
2. in our case Application Layer send packet on sx1276 but sx1276 not transmit packet Over the Air

What is issue behind that can you please help me

Thank you

mluis1, 2019-03-29:
I close this issue as it looks like you have already solved the original issue.

closed 2019-03-29 by mluis1