Lora-net/LoRaMac-node #674: Cmake Fails Compilation on Ubuntu

adityaseven, 2019-02-08:
I followed the instructions given in 
https://github.com/Lora-net/LoRaMac-node/wiki/Development-environment

When I ran the command 
`cmake -DCMAKE_TOOLCHAIN_FILE="cmake/toolchain-arm-none-eabi.cmake" ..`

I get the following output
"
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/bin/arm-none-eabi-gcc" is not able to compile a simple
  test program.

  It fails with the following output:

   Change Dir: /home/ANT.AMAZON.COM/renukr/workspace/work/ring/LoRaMac-node/build/CMakeFiles/CMakeTmp

  

  Run Build Command:"/usr/bin/make" "cmTC_4ecb5/fast"

  /usr/bin/make -f CMakeFiles/cmTC_4ecb5.dir/build.make
  CMakeFiles/cmTC_4ecb5.dir/build

  make[1]: Entering directory
  '/home/ANT.AMAZON.COM/renukr/workspace/work/ring/LoRaMac-node/build/CMakeFiles/CMakeTmp'

  Building C object CMakeFiles/cmTC_4ecb5.dir/testCCompiler.c.obj

  /usr/bin/arm-none-eabi-gcc -o CMakeFiles/cmTC_4ecb5.dir/testCCompiler.c.obj
  -c
  /home/ANT.AMAZON.COM/renukr/workspace/work/ring/LoRaMac-node/build/CMakeFiles/CMakeTmp/testCCompiler.c

  Linking C executable cmTC_4ecb5

  /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_4ecb5.dir/link.txt
  --verbose=1

  /usr/bin/arm-none-eabi-gcc CMakeFiles/cmTC_4ecb5.dir/testCCompiler.c.obj -o
  cmTC_4ecb5

  
  /usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o):
  In function `exit':

  exit.c:(.text.exit+0x2c): undefined reference to `_exit'

  collect2: error: ld returned 1 exit status

  CMakeFiles/cmTC_4ecb5.dir/build.make:97: recipe for target 'cmTC_4ecb5'
  failed

  make[1]: *** [cmTC_4ecb5] Error 1

  make[1]: Leaving directory
  '/home/ANT.AMAZON.COM/renukr/workspace/work/ring/LoRaMac-node/build/CMakeFiles/CMakeTmp'

  Makefile:126: recipe for target 'cmTC_4ecb5/fast' failed

  make: *** [cmTC_4ecb5/fast] Error 2

  

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt

-- Configuring incomplete, errors occurred!
See also "/home/ANT.AMAZON.COM/renukr/workspace/work/ring/LoRaMac-node/build/CMakeFiles/CMakeOutput.log".
See also "/home/ANT.AMAZON.COM/renukr/workspace/work/ring/LoRaMac-node/build/CMakeFiles/CMakeError.log".
"
Attached CMakeError.log and CMakeOutput.log

[CMakeError.log](https://github.com/Lora-net/LoRaMac-node/files/2847140/CMakeError.log)
[CMakeOutput.log](https://github.com/Lora-net/LoRaMac-node/files/2847141/CMakeOutput.log)

ConnyBusy, 2019-02-09:
Prerequisite from https://github.com/Lora-net/LoRaMac-node/wiki/Development-environment
CMake >= 3.6 
You are using CMake == 3.5
It works on my end with Ubuntu with out errors.  I have cmake version 3.10.2 on my machine Ubuntu 18.04 LTS Server

mluis1, 2019-02-13:
Can we close this issue?

adityaseven, 2019-02-15:
Hah I missed that. It works now!

closed 2019-02-15 by adityaseven