Lora-net/LoRaMac-node #679: SeNvmCtxChanged( ); STUCK

PellegrinoA, 2019-02-18:
Hi all!
In soft-se.c the fuction SeNvmCtxChanged( ); is stucked.
Any solution?

SecureElementStatus_t SecureElementSetKey( KeyIdentifier_t keyID, uint8_t* key )
{
    if( key == NULL )
    {
        return SECURE_ELEMENT_ERROR_NPE;
    }
    SecureElementStatus_t retval = SECURE_ELEMENT_ERROR;

    for( uint8_t i = 0; i < NUM_OF_KEYS; i++ )
    {
        if( SeNvmCtx.KeyList[i].KeyID == keyID )
        {
            if( LORAMAC_CRYPTO_MULITCAST_KEYS < SeNvmCtx.KeyList[i].KeyID )
            {  // Decrypt the key if its a Mulitcast key

                uint8_t decryptedKey[16] = { 0 };

                retval = SecureElementAesEncrypt( key, 16, MC_KE_KEY, decryptedKey );
                if( retval != SECURE_ELEMENT_SUCCESS )
                {
                    return retval;
                }
            }
            else
            {
                memcpy1( SeNvmCtx.KeyList[i].KeyValue, key, KEY_SIZE );
                **SeNvmCtxChanged( );**
                return SECURE_ELEMENT_SUCCESS;
            }
        }
    }

closed 2019-02-18 by PellegrinoA

Mani-Sadhasivam, 2019-10-28:
@PellegrinoA Did you happen to figure out the issue? I'm experiencing the same...

Mani-Sadhasivam, 2019-10-28:
@mluis1 Any clue about this one?

C47D, 2019-11-08:
@Mani-Sadhasivam I have the same issue, the code is in the infinite while loop of the `HardFault_Handler()`, not stuck per se.

Some information of my application configuration:
BOARD = NucleoL476
APPLICATION=LoRaMac-classA
REGION=REGION_US915

Mani-Sadhasivam, 2019-11-15:
@C47D Thanks a lot! Yes, it turned out to be the same issue as yours and now it's fixed.

C47D, 2019-11-15:
@Mani-Sadhasivam, no problem, now (HEAD) the examples validates the return from `LoRaMacInitialization()`

```c
status = LoRaMacInitialization( &macPrimitives, &macCallbacks, ACTIVE_REGION );
if ( status != LORAMAC_STATUS_OK )
{
    printf( "LoRaMac wasn't properly initialized, error: %s", MacStatusStrings[status] );
    // Fatal error, endless loop.
    while ( 1 )
    {
    }
}
```