Lora-net/LoRaMac-node #681: SX126xIoInit() not called from SX126xInit()

5frank, 2019-02-18:
Why is sx126x-board.c:SX126xIoInit() not called from from sx126x.c:SX126xInit()? This causes higher abstractions layers to depend on the exact chip model. Isn't the purpose with the radio.h interface to hide details like this?

nestorayuso, 2019-02-18:
a specific board is dependant to a selected radio.

when you design a new PCB board you should select a transceiver and a collection of GPIO pins and a SPI interface.

5frank, 2019-02-18:
Yes, but isn't the objective to always abstract the hardware when possible? In this case I seems to ruin the abstraction model provided by radio.h for no apparent reason (as far as I can tell). Nothing is gained, only makes porting to new hardware more complicated.

mluis1, 2019-03-22:
If you have a closer look at the `board.c` file you will notice that the radio `SPI` as well as the radio `GPIO`s are initialized/de-initialized at boot and every time the MCU goes to/from a low power mode.

The board file is platform specific and we beeleave that it doesn't break the abstraction model.

mluis1, 2019-04-02:
Can we close this issue?

5frank, 2019-04-03:
I surrender.

closed 2019-04-03 by 5frank