Lora-net/LoRaMac-node #684: keil

megahertzmetal, 2019-02-24:
can you add support keil? or maybe you have old project on the keil ?

djaeckle, 2019-03-01:
Hi megahertzmetal,

we have sample projects for old versions of this repository. Please take a look e.g. at tag v4.0.0.

closed 2019-03-01 by djaeckle

yanzi418, 2019-03-02:
I wrote a lorawan code for keil project based on stm32l073rz

https://download.csdn.net/download/qq_34710245/10982357

megahertzmetal, 2019-03-03:
yanzi418, thank you! but i can't download your project because i'm from europe region. Maybe you can upload project on other resource? maybe https://dfiles.ru or dropbox?

megahertzmetal, 2019-03-03:
and i have questions, why u use STM32L151 platform for your project? why not F series?

yanzi418, 2019-03-05:
megahertzmetal Give me your mailbox I'll send it to you.

yanzi418, 2019-03-05:
megahertzmetal，My mailbox is 975375246@qq.com