Lora-net/LoRaMac-node #685: MLME Confirm: Tx timeout

sagar-patel-sls, 2019-02-26:
Hii, @mluis1 
I got Below error while debugging code of [feature/5.0.0](https://github.com/Lora-net/LoRaMac-node/tree/feature/5.0.0) branch
```
###### ===== MLME-Confirm ==== ######
STATUS      : Tx timeout
```
[image]

Please help me..
Thanks

sagar-patel-sls, 2019-03-25:
Hii, All
Anyone can help me??

mluis1, 2019-03-28:
A `TxTimeout` should never happen. Once the radio is asked to transmit something the execution should always finalize by a `TxDone` interrupt.

That said, please find below some reasons which could cause the issue that you are observing:

1. The radio chip is dead for some reason. Have you tried with another chip?
2. The connections between the radio chip and the micro-controller are not OK. Please verify that all connections are OK. For instance if the `TxDone` IRQ pin is not connected then you will get this behavior.
3. Something that can cause this would be if we have made a mistake defining the micro-controller pins connected to the radio. Could you please verify the file `board-config.h` for your board?
We don't test all the possible combinations every day so it can happen that we make an unwanted change.
4. Sometimes if the PCB is not well designed the SPI communication can get corrupted that's why in the function `SX127xOnTimeoutIrq` there is a workaround implemented.
5. Have you tried to run one of the provided examples without any changes?

We can't do anything else to help you.

sagar-patel-sls, 2019-03-29:
Hii @mluis1 

**1.** The radio chip is dead for some reason. Have you tried with another chip? **-->** I have tried with another chip 
**5.** Have you tried to run one of the provided examples without any changes? **-->** I have tried to run one of the provided examples but it didn't works, v1.0.3 branch it was successfully works

Thank you

mluis1, 2019-03-29:
Unfortunately I have no spare time to perform such task.

I would recommend that you take a look at the Nucleo and Shield schematics and verify your `board-config.h` file.

If you don't mind I close this issue as we can't do much more to help you.

closed 2019-03-29 by mluis1