Lora-net/LoRaMac-node #686: The node can't go to sleep when slowing down the system clocks

outely, 2019-02-27:
hello!

Recently, I found such a problem: when the node works at 32MHz by default, it can enter the low power mode normally. However, in order to reduce the power consumption during operation, when I reduce the system frequency below 4MHz, the node can't enter the low power mode. However, when the system is running above 6MHz, it can enter the low power mode normally. Has anyone tried to sleep normally below 4MHz?

mluis1, 2019-04-18:
This question is not related to this project it is more a question on how to use the MCU. 
We would recommend that you ask your question directly to your MCU provider.

closed 2019-04-18 by mluis1

fahad80, 2019-04-25:
@outely I have tried MSI at 2.1MHz with STM32L053 and was able to go to the stop mode.

outely, 2019-04-26:
@fahad80 Thank you for your reply. I used HSI before. Now I try MSI and see if the MCU can enter sleep mode.