Lora-net/LoRaMac-node #689: How to import this source in atollic studio

nageshdivatagi, 2019-03-01:
I am beginner of this how to import this source in atollic studio  or else can provide atollic true studio source code

trick2011, 2019-03-20:
I have no idea myself but compilation of the code can be achieved by following the following instructions: https://github.com/Lora-net/LoRaMac-node/blob/develop/Doc/development-environment.md. By setting it up this way you might get to understand how to convert it into a atollic studio project. You are of course free to add these instructions to the documentation afterwards

fahad80, 2019-04-02:
I have imported the previous project (early in 2018) to atollic truestudio. Couple of things to follow.
1) Create a dummy project with right mcu using STM32CubeMx. CubeMx has TrueStudio support.
2) Update "Include".
3) Update "Source".
4) Filter unnecessary files in the "Source".
5) Add a pre-processor for the region you are interested in.

mluis1, 2019-04-18:
Can we close this issue?

nageshdivatagi, 2019-04-18:
Have you tried this source?

mluis1, 2019-04-18:
What do you mean?

There is nothing that we can do for you.

Have you tried the tips provided by @trick2011 and @fahad80 ?

SuadM5, 2019-04-20:
> I have imported the previous project (early in 2018) to atollic truestudio. Couple of things to follow.
>
> 1. Create a dummy project with right mcu using STM32CubeMx. CubeMx has TrueStudio support.
> 2. [...]

Hi @fahad80,

I am in the same situation. I can't figure out the CMake or how to set it up and I have TrueStudio and STM32CubeMx already set up. 

Can you be more specific please. Do you mean that I should copy all the .h files in the LoRaMAC code to Inc, and all the .c ones to Src? But then How can I actually link it to my main with the right hardware setup?

I have tried using I-cube-lrwan and the generated CubeMx files where I copied one of the examples' main.c file to try and implement my project. But I got a lot of errors and I read that this is not really an option as the i-cube-lrwan is not made to be used as a middleware. 

Now I'm trying to use the files in this repository and I am really lost. 

Is there a video or something online on how to setup the environment?

PS I am using RF_LORA_SO which has SX1272 transceiver and an STM32L072CZ MCU. 

Thanks,

selimsagir, 2019-04-23:
> I have imported the previous project (early in 2018) to atollic truestudio. Couple of things to follow.
>
> 1. Create a dummy project with right mcu using STM32CubeMx. CubeMx has TrueStudio support.
> 2. [...]

Did you changed Drivers files with source codes?

fahad80, 2019-04-25:
@SuadM5 I prepared an Atollic TrueStudio project using the latest I-CUBE-LRWAN (1.2.1) and STM32L072
https://github.com/fahad80/ICubeLorawan_TrueStudio

SuadM5, 2019-04-25:
@fahad80 Thanks a lot, I will give it a go.

selimsagir, 2019-04-25:
> @SuadM5 I prepared an Atollic TrueStudio project using the latest I-CUBE-LRWAN (1.2.1), STM32L072 and SX1272.
> https://github.com/fahad80/ICubeLorawan_TrueStudio
> 
> I converted an End Node example.
> 
> N.B. I-Cube-Lrwan has examples for 'System Workbench for STM32'. TrueStudio and SW4ST both are based on eclipse. So, you can give that a try.

Do you have a project with LoRaMac-node and CubeMX? By the way, thanks for your answer.

fahad80, 2019-04-26:
@selimsagir Didn't understand your question. The repo I shared uses CubeMX hardware library. But what do you mean by LoRaMac-node?

nageshdivatagi, 2019-04-26:
thank you so much sir..

selimsagir, 2019-04-26:
@fahad80 Oh, i see. Thanks for your information. I have edited your example according to my kit which is B-L072Z-LRWAN1. I am trying to make an example with an LCD and GPS. So, can I make it done with your repo?

fahad80, 2019-04-28:
@selimsagir I believe you can. For example, if you use Ublox GPS M8, they have UART interface. Just use UART library from CubeMx. I have done that.

selimsagir, 2019-04-30:
@fahad80 Thanks for your replies. I will try that, i hope i can :)

closed 2019-06-07 by mluis1