Lora-net/LoRaMac-node #692: Nucleo L073Z Sleep Current is High

matomback, 2019-03-11:
Hi,

I am using the class A example on the Nucleo L073Z board and seeing around 4.4 mA when in sleep mode between transactions, up to around 7mA during wakeup and transmit. When I implemented low power sleep instead I was able to reduce to 3.3 mA when sleeping but no lower.

I am using USB power through laptop, have added an I2c sensor that is polled when the device wakes, implemented the NVM management to retain keys, but other than that have not really changed much.

I am struggling to figure out the issue here. Could this be related to the USE_DEBUGGER definition? Neither changing this definition or switching to Release build seems to solve the problem. Anything else related to the stack or STM32 Nucleo boards that needs to be modified/checked in order for sleep modes to function correctly? Any information pointing me in the right direction would be greatly appreciated.

Thanks

jamesl-dm, 2019-03-12:
Did you remember to power cycle your board after disconnecting the debugger, and before testing? A lot of micros will spin up auxiliary debug clocks when you attach the debugger, and they'll stay running until the next power cycle. They usually draw on the order of 3 mA.

matomback, 2019-03-18:
Thank you for the feedback. Yes, I power cycle the board after programming and before testing power draw. I am using VS Code IDE to program and I have also tried programming without the debugger - via the release build or setting the USE_DEBUGGER option to OFF. With the release build the debugger appears to still be attached. I tried to run the CMake Command "Execute the target without the debugger" but it throws an error. With USE_DEBUGGER option set to OFF in CMakeLists.txt within the src folder, nothing seems to be changed and the debugger still runs. 

I also discovered that going to stopmode instead of sleep/lowpowersleep seems to work correctly and I was able to get the current draw to drop to expected values, however I am still getting  ~ 4.4mA/3.3mA when in sleep/lowpowersleep respectively.

pdemil, 2019-03-20:
stopmode is not used when using USB power.

matomback, 2019-03-21:
I made the device enter stopmode between uplinks while on USB power for testing purposes. Yes I found that stopmode does turn off USB power and ST LINK functionality. It worked as far as getting the board to sleep with expected low current draw when I was testing sensor readings prior to joining. But it shuts everything down and the device will not stay joined or rejoin correctly when it wakes back up. I imagine this would be due to the USB power, as I'd expect the device can stay joined between uplinks when it goes into stopmode.

So this still does not solve my issue with high current when in sleep mode & on USB power. Is there a way within the stack or VS Code IDE to ensure that the debugger is set off and the debug clocks do not run on ~ 3mA while the device is sleeping (even after power-cycling as jamesl-dm suggested)? Has anyone else experienced this issue using the STM Nucleo boards and sx1262 radio?

fahad80, 2019-04-02:
In stop mode the data in ram is retained. So the device should stay joined. When you say your board is USB powered, did you mean the power come from a usb port to your mcu's vcc? In that case, your device should be still powered during stop mode. Mcu cannot turn off that USB power.

I implemented stop mode for L053 though, and usually power my device from usb. Didn't have an issue.

closed 2019-06-07 by mluis1