Lora-net/LoRaMac-node #693: which version of GNU ARM-Toolchain is required

trick2011, 2019-03-13:
In the prerequisites of the development environment for cmake a version is stated but not for GNU ARM-Toolchain. It is said that the version on 16.04 too old is but no mention is made of the 18.04 version. Can the minimum required GNU ARM-Toolchain version be added to the list?
[prequisites](https://github.com/Lora-net/LoRaMac-node/wiki/Development-environment#prerequisites)

temporaryaccount, 2019-03-13:
Ubuntu 18.04's toolchain works fine, just make sure to manually update `libnewlib` until the bug is fixed.

bbcmicrobit/micropython#514

https://bugs.launchpad.net/ubuntu/+source/gcc-arm-none-eabi/+bug/1767223

trick2011, 2019-03-15:
I added the information to the documentation and created a pull request. I think this thread can be closed.

#695

closed 2019-03-15 by mluis1