Lora-net/LoRaMac-node #699: class-C RX2 window not working anymore?

gpgib6, 2019-03-19:
Hi, I haven't updated the LoRa stack in a while, I'm using 5.0.0 feature branch. Today I did an update and now the RX2 window seems to be broken.

Before the update, I can go to the MultiTech gateway and send a downlink at any time, the downlink is sent out right away and the device picks it up, no problems.
Now after the update, same thing, I send the downlink, it is sent out right away but the device pretends like nothing is going on, as if the continuous RX2 window didn't exist anymore.

What do you think went wrong? Is there anything I can check?
For the time being I went back to the previous version of the stack but here I'm dealing with the rare condition where the lora stack gets stuck, thinking it's in TX mode.

mluis1, 2019-03-22:
Could you please provide the commit hash that is working as expected?
Could you please provide which example, platform that you are using?

We will provide a feedback as soon as possible

yanzi418, 2019-03-25:
I also had similar problems with lorawan 1.03 development.

static void OpenContinuousRxCWindow( void )
{
    MacCtx.RxWindowCConfig.RxSlot = RX_SLOT_WIN_CLASS_C;
    // Setup continuous listening
    MacCtx.RxWindowCConfig.RxContinuous = true;
	
	//add
	MacCtx.RxWindowCConfig.Channel = MacCtx.Channel;
	MacCtx.RxWindowCConfig.Frequency = MacCtx.NvmCtx->MacParams.RxCChannel.Frequency;
	MacCtx.RxWindowCConfig.DownlinkDwellTime = MacCtx.NvmCtx->MacParams.DownlinkDwellTime;
	MacCtx.RxWindowCConfig.RepeaterSupport = MacCtx.NvmCtx->RepeaterSupport;
	MacCtx.RxWindowCConfig.RxSlot = RX_SLOT_WIN_CLASS_C_MULTICAST;
	MacCtx.RxWindowCConfig.RxContinuous = true;
	
    // At this point the Radio should be idle.
    // Thus, there is no need to set the radio in standby mode.
    if( RegionRxConfig( MacCtx.NvmCtx->Region, &MacCtx.RxWindowCConfig, ( int8_t* )&MacCtx.McpsIndication.RxDatarate ) == true )
    {
        Radio.Rx( 0 ); // Continuous mode
        MacCtx.RxSlot = MacCtx.RxWindowCConfig.RxSlot;
    }
}

yanzi418, 2019-03-25:
Set to CLASS_C device, but the second receiving window is not configured

yanzi418, 2019-03-25:
I set the demo routine to CLASS_C, but MacCtx. Rx Windows CConfig. Frequency is 0

Mr-Techtron, 2019-04-19:
Has anybody found why the frequency is set to 0?

gpgib6, 2019-05-07:
I'm using the 5.0.0 branch and this function gets called: void SX1276SetChannel( uint32_t freq )
The frequency is also set to 0 when it should be 923300000 or RX1 window I believe.

khomplora, 2019-05-29:
I have the same problem :/

yanzi418, 2019-05-30:
The code I posted can solve this problem temporarily.

yanzi418, 2019-05-30:
In OpenContinuousRxCWindow function，add
//add
MacCtx.RxWindowCConfig.Channel = MacCtx.Channel;
MacCtx.RxWindowCConfig.Frequency = MacCtx.NvmCtx->MacParams.RxCChannel.Frequency;
MacCtx.RxWindowCConfig.DownlinkDwellTime = MacCtx.NvmCtx->MacParams.DownlinkDwellTime;
MacCtx.RxWindowCConfig.RepeaterSupport = MacCtx.NvmCtx->RepeaterSupport;
MacCtx.RxWindowCConfig.RxSlot = RX_SLOT_WIN_CLASS_C_MULTICAST;
MacCtx.RxWindowCConfig.RxContinuous = true;

mluis1, 2019-06-17:
Is this issue happening in both ABP and OTAA activation modes?

From our analysis it should only happen in ABP mode.

yanzi418, 2019-06-17:
I only tested it in OTAA.

mluis1, 2019-06-17:
Ok, thanks for the feedback.

Mr-Techtron, 2019-06-17:
And I have tested in ABP mode.

mluis1, 2019-06-17:
@yanzi418 and @Mr-Techtron, Could you please check that the below patch solves the observed issue?
As each one is using a different activation method this will speedup the verification.

On my side I will check that the FUOTA project is still operating as expected.

Thanks in advance.

```patch
diff --git a/src/mac/LoRaMac.c b/src/mac/LoRaMac.c
index 319a86fb..145bcd47 100644
--- a/src/mac/LoRaMac.c
+++ b/src/mac/LoRaMac.c
@@ -2586,6 +2586,14 @@ static void ResetMacParameters( void )
     // Initialize channel index.
     MacCtx.Channel = 0;
     MacCtx.NvmCtx->LastTxChannel = MacCtx.Channel;
+
+    // Initialize Rx2 config parameters.
+    MacCtx.RxWindow2Config.Channel = MacCtx.Channel;
+    MacCtx.RxWindow2Config.Frequency = MacCtx.NvmCtx->MacParams.Rx2Channel.Frequency;
+    MacCtx.RxWindow2Config.DownlinkDwellTime = MacCtx.NvmCtx->MacParams.DownlinkDwellTime;
+    MacCtx.RxWindow2Config.RepeaterSupport = MacCtx.NvmCtx->RepeaterSupport;
+    MacCtx.RxWindow2Config.RxContinuous = false;
+    MacCtx.RxWindow2Config.RxSlot = RX_SLOT_WIN_2;
 }
 
 /*!
```

khomplora, 2019-06-17:
> @yanzi418 and @Mr-Techtron, Could you please check that the below patch solves the observed issue?
> As each one is using a different activation method this will speedup the verification.
> 
> On my side I will check that the FUOTA project is still operating as expected.
> 
> Thanks in advance.
> 
> ```diff
> diff --git a/src/mac/LoRaMac.c b/src/mac/LoRaMac.c
> index 319a86fb..145bcd47 100644
> --- a/src/mac/LoRaMac.c
> +++ b/src/mac/LoRaMac.c
> @@ -2586,6 +2586,14 @@ static void ResetMacParameters( void )
>      // Initialize channel index.
>      MacCtx.Channel = 0;
>      MacCtx.NvmCtx->LastTxChannel = MacCtx.Channel;
> +
> +    // Initialize Rx2 config parameters.
> +    MacCtx.RxWindow2Config.Channel = MacCtx.Channel;
> +    MacCtx.RxWindow2Config.Frequency = MacCtx.NvmCtx->MacParams.Rx2Channel.Frequency;
> +    MacCtx.RxWindow2Config.DownlinkDwellTime = MacCtx.NvmCtx->MacParams.DownlinkDwellTime;
> +    MacCtx.RxWindow2Config.RepeaterSupport = MacCtx.NvmCtx->RepeaterSupport;
> +    MacCtx.RxWindow2Config.RxContinuous = false;
> +    MacCtx.RxWindow2Config.RxSlot = RX_SLOT_WIN_2;
>  }
>  
>  /*!
> ```

Work for me! (ABP)

mluis1, 2019-06-28:
We assume that the OTAA is also working as there is no feedback. We will apply this patch as soon as possible.

closed 2019-07-02 by mluis1

Pirakutu, 2019-07-24:
Hi there, this is still not working correctly. For example when the device received a new set of RX2/RXC params per MAC command, these changes are saved into the NVM variables but not into the actual running variables. This is not too bad for the RX2 params, as these are set once the timer event kicks in after sending a message. However, the local RXC params are never updated. I added this to ScheduleTX in LoRaMac.c to solve the problem (the reason i added it there is that the servers usually change the devices RX2/RXC frequecies once they have received an answer from the node, as this is a ACK MAC command [triggers a tx needed event]):

    // Compute Rx1 windows parameters
    RegionComputeRxWindowParameters( MacCtx.NvmCtx->Region,
                                     RegionApplyDrOffset( MacCtx.NvmCtx->Region, MacCtx.NvmCtx->MacParams.DownlinkDwellTime, MacCtx.NvmCtx->MacParams.ChannelsDatarate, MacCtx.NvmCtx->MacParams.Rx1DrOffset ),
                                     MacCtx.NvmCtx->MacParams.MinRxSymbols,
                                     MacCtx.NvmCtx->MacParams.SystemMaxRxError,
                                     &MacCtx.RxWindow1Config );
    // Compute Rx2 windows parameters
    RegionComputeRxWindowParameters( MacCtx.NvmCtx->Region,
                                     MacCtx.NvmCtx->MacParams.Rx2Channel.Datarate,
                                     MacCtx.NvmCtx->MacParams.MinRxSymbols,
                                     MacCtx.NvmCtx->MacParams.SystemMaxRxError,
                                     &MacCtx.RxWindow2Config );

    /* ADDED CODE */
    if( ( MacCtx.NvmCtx->DeviceClass == CLASS_C ) && ( MacCtx.NvmCtx->NetworkActivation != ACTIVATION_TYPE_NONE ) )
    {
        // We can only compute the RX C window parameters, if we are already
        // in class c mode and joined. We cannot setup an RX C window in case of any other
        // class type.
        // Update RxC frequency if it has changed
        if (MacCtx.RxWindowCConfig.Frequency !=  MacCtx.NvmCtx->MacParams.RxCChannel.Frequency){
          MacCtx.RxWindowCConfig.Frequency = MacCtx.NvmCtx->MacParams.Rx2Channel.Frequency;
        }
        // Compute RxC windows parameters if datarate has changed
        if (MacCtx.RxWindowCConfig.Datarate != MacCtx.NvmCtx->MacParams.RxCChannel.Datarate)
        RegionComputeRxWindowParameters( MacCtx.NvmCtx->Region,
                                         MacCtx.NvmCtx->MacParams.RxCChannel.Datarate,
                                         MacCtx.NvmCtx->MacParams.MinRxSymbols,
                                         MacCtx.NvmCtx->MacParams.SystemMaxRxError,
                                         &MacCtx.RxWindowCConfig );
    }
    /* ADDED CODE */