Lora-net/LoRaMac-node #704: Generic GNU Toolchain / Atollic TrueSTUDIO support

ponyatov, 2019-03-27:
What is the right way to organize files inside LoRaMac-node project if I want to adapt it to use with GNU toolchain and free tools?

What I want to add:
- generic arm-none-eabi GNU toolchain available in most Linux distros
- Atollic TrueSTUDIO & generic Eclipse IDEs (Makefile-based project)
- add two external SX127x boards (AcSiP S76 and EBYTE E19-868Mx0S)

ponyatov, 2019-03-27:
I see `.vscode` directory, is it legal to put a few `.*project` files in the root directory, which stores IDE some configuration I want to share with everybody?

mluis1, 2019-04-18:
This project already uses GNU compiler and provides required files to use `cmake` which will generate the `makefiles` required to build the project.

It also provides instructions on how to used VSCode and KDevelop as IDE. All these are "free tools".

Please refer to the following document [development-environment.md](https://github.com/Lora-net/LoRaMac-node/blob/develop/Doc/development-environment.md)

In your fork you can do whatever pleases you.

As you can understand we can not add support for every sort of IDE available on the market.

closed 2019-04-18 by mluis1