Lora-net/LoRaMac-node #705: Class C don't start

PellegrinoA, 2019-03-27:
Hi! Anyone knows How class C start?
Because when It start in console appear:

###### ===== MLME-Request - MLME_JOIN ==== ######
STATUS      : Duty-cycle restricted

###### ===== MLME-Indication ==== ######
STATUS      : OK

This was repit all time and send one after another without wait nothing.

Any solution or idea?

Thank you!

mluis1, 2019-03-28:
From the short description and log that you provide it looks like that the node hasn't yet joined the network.

The `STATUS : Duty-cycle restricted` is shown because the node already tried to join the network without success. As per LoRaWAN specification there are duty-cycle restrictions on the Join procedure.
If you let the node run for some more time you will notice that a JoinReq will be sent as soon as the restriction doesn't apply anymore.

This means that up until the node successfully joins the network there is nothing you can do or state.

My advise is that you verify your activation parameters (AppEUI, AppKey, DevEUI) on the node as well as on the network server.

Which region are you using?

PellegrinoA, 2019-03-28:
Hi! Thank for answer.
But my node have ABP activation and it use AU_915_918.
I don't know why my nodo (class C) send a JoinReq if it is in ABP. I'm confuse.

mluis1, 2019-03-29:
For some reason your implementation is sending join requests.

As we don't know which modifications you have performed on the code we can't help you much more.

Could you please try one of the examples provided by this project?

Please try it without doing modifications. By default the projects provided here are setup to use ABP activation.

mluis1, 2019-04-18:
Can we close this issue?

closed 2019-04-18 by PellegrinoA