Lora-net/LoRaMac-node #706: How to change Spreading Factor (SF)

danielrsfreitas, 2019-03-27:
Hello,

In my country (Brazil) the regulation says the data can not stay on air for more than 400 ms. 

Does anyone know how to change the Spreading Factor (SF) from SF12 to a lower SF, let's say SF10 and help me understand what changes in behavior will be related to this modification ?

ConnyBusy, 2019-03-28:
Hi @danielrsfreitas 
I suppose you are using EU433 Band. You can do this by changing the following in RegionEU433.h File 

**FROM** 
[image]

**TO**
[image]

danielrsfreitas, 2019-03-28:
Hi @ConnyBusy ,

I'm actually using AU915 band. I already changed DR_0 to DR_2 but it didn't seem to work. The only thing that really changed the SF was:

[image]

But I'm not sure if this is the best.

Thanks!

mluis1, 2019-03-28:
Without changes the AU915 region is already compliant with your requirements.
If you look at file `RegionAU915.h` you will see the bellow line

```c
#define AU915_DEFAULT_UPLINK_DWELL_TIME             1
```
When set to 1 it means that the 400 ms restrictions are enabled.

This means that by default the minimum usable data rate is DR_2 which ensures that the 400 ms restrictions are applied.

danielrsfreitas, 2019-03-28:
Hi @mluis1 ,

Just saw your commit "e60bd869 Update AU915 and US915 regions according to "LoRaWAN Regional Parameters v1.0.3rA" specification."

I realized that I'm using release 4.4.1.

Thank you so much!

closed 2019-03-28 by danielrsfreitas