Lora-net/LoRaMac-node #707: MAC.Command.DeviceTimeAns - 18 seconds in future

branek, 2019-03-28:
On develop/5.0.0 branch, when we send `DeviceTimeReq` MAC command, and server answers back, time received is 18 seconds in future.

Stack is tested against 4 major network servers implementation, one does not support request, and three other are sending answer with 18 seconds in future.

We did not take to much time to investigate issue, but it's probable related to the GPS/UTC leap seconds handling. [Here](https://github.com/brocaar/loraserver/blob/1a3144370fc698f5c5dadd0235bcb75a9fbb9f26/internal/gps/gps.go) is implementation as done by loraserver.io.

Is there any suggestion how this should be corrected on device side?

ConnyBusy, 2019-03-28:
is this issue not related to issue #552

mluis1, 2019-03-28:
It is now sometime that we didn't test our self this implementation against the major network providers but the last time we did the provided times were good.

Please also note that the node implementation doesn't handle the leap seconds as there is no standard way to provide this value to the node.
All timings in the node implementation are based on the GPS epoch time. In the node when the time is converted to Unix time we just apply the `UNIX_GPS_EPOCH_OFFSET` which has a value of `315964800`

Since a few days now I'm testing the loraserver.io (docker-compose, latest version) setup for LoRaWAN 1.1.1-ClassB with feature/5.0.0 also setup for LoRaWAN 1.1.1 (EU868) and the received Ans looks to be correct.

What I'm observing is that the time provided by the server has some randomness, between 0 and 1 second around the expected beacon transmission time.

I'm trying to figure out if the randomness is introduced by the server or by the node computations.

To show you my observations, I have just reset my node and made some oscilloscope screenshots up until the node switches to ClassB.

To produce the screenshots I have enable the "USE_RADIO_DEBUG":"ON" on settings.json file.
The signals shown on the screenshots are:
D7: Radio Tx debug pin
D6: Radio Rx debug pin
3  : Gateway Tx - Oscilloscope probe is connected to the gateway RF output and set to 50 Ohms impedance.

scope_0.png: 1st attempt to acquire the beacon FAILED.
scope_1.png: 2nd attempt to acquire the beacon FAILED.
scope_2.png: 3rd attempt to acquire the beacon FAILED.
scope_3.png: 4th attempt to acquire the beacon FAILED.
scope_4.png: 5th attempt to acquire the beacon SUCCESS.
scope_5.png: node switches to ClassB, we can see the ping-slots being open every second.
scope_6.png: 1st beacon reception after switching to ClassB.
[class-b-beacon-acquisition.zip](https://github.com/Lora-net/LoRaMac-node/files/3019235/class-b-beacon-acquisition.zip)

branek, 2019-04-01:
We do not use Class B right now, and after some search, we are able to understand time represented in GPS. Good source of informations is [LeapSecond](http://leapsecond.com/java/gpsclock.htm).

As shown, GPS time is indeed 18 seconds ahead of UTC at this moment. And, as you already pointed out, this time gap is not constant, and it's not easy to do conversion to UTC on the node. 

Thank you for clarifying this.

So, first post should be more detailed in explanation, something like "18 seconds ahead of UTC". Topic changed to better fit question.

mluis1, 2019-04-02:
As I understand we can then close this issue. Isn't it?

branek, 2019-04-02:
That's correct. Thanks.

closed 2019-04-02 by mluis1