Lora-net/LoRaMac-node #709: Next Bacon Channel poorly calculated (AU_915)

PellegrinoA, 2019-04-01:
Hi All!!!

I'm prove the class B in AU_915. But the Beacon channel (in the node) is not the channel that the gateway use.
Also the timing is wrong because the windows is opened after the gateway send a beacon.

Some solucion or advice???
Thank!

ConnyBusy, 2019-04-01:
Hi @PellegrinoA,

There is an issue concerning beacon acquisition at startup caused by device time request/answer. #552 
Maybe you let you Node acquire a beacon before you configure you server to process uplinks from the Node. This a workaround for me which enables me work correctly with Class B

Per source code:
[image]
and specification:
[image]
If there is an issue with beacon time then the channels will not match as well.

PellegrinoA, 2019-04-02:
Hi @ConnyBusy !

The configuration is ok! 

Now, I see the timeSinceGPSEpoch (DeviceTimeReq "Downlink") in the server:

[image]

But the code that save the timeSinceGPSEpoch in the device have only 32 bits and the server use more than 32bits:
                gpsEpochTime.Seconds = ( uint32_t )payload[macIndex++];
                gpsEpochTime.Seconds |= ( uint32_t )payload[macIndex++] << 8;
                gpsEpochTime.Seconds |= ( uint32_t )payload[macIndex++] << 16;
                gpsEpochTime.Seconds |= ( uint32_t )payload[macIndex++] << 24;
                gpsEpochTime.SubSeconds = payload[macIndex++];

I did a print and obtain:
sec= 1238212767 / subsec= 34
Both are different.
Where is the wrong???

ConnyBusy, 2019-04-02:
Hi @PellegrinoA,
Yes, the code saves it correctly. The DeviceTimeAns payload is all together 40 bits ( 5Bytes) with 32 bits carrying the seconds and 8 bits carrying the fractional-seconds (SubSeconds) or better still milliseconds as the code above does. 
[image]

Your server timeSinceGPSEpoch doesn't really look right to me. It is too long. The fractional seconds must only be 8 bits in size nothing more

mluis1, 2019-04-05:
We are currently testing the fix for #552 using loraserver.io setup for ClassB EU868.

@PellegrinoA the provided image looks to be from loraserver.io isn't it.
From our tests the loraserver.io is providing the right DeviceTimeAns.

closed 2019-04-05 by mluis1

GuntherSchulz, 2019-04-05:
Tomorrow 6 April is GPS Epoch Rollover.

How is LoRaWan affected?

How is this codebase affected?

mluis1, 2019-04-05:
The rollover as I understand it only concerns the week number.

The LoRaWAN protocol and this code base only rely on the number of seconds elapsed since the GPS epoch thus, they aren't concerned by this situation.

The number of seconds since GPS epoch will rollover on 
`2^32 seconds -> 4294967296 seconds -> Feb 12, 2116 at 06:27:58 UTC time.`