Lora-net/LoRaMac-node #713: Doesn't work in latest toolchain

Gmlmk1, 2019-04-04:
Hi,
I'm currently  working on latest release (4.4.1, visual code,sk-im880b). 
So, if i try compile project with arm toolchain 7.3.0 then board doesn't work. But the program works in 4.9.3 version. 
What could be the problem? In the code did not change anything.

mluis1, 2019-04-10:
Unfortunately, you don't provide enough information to help you.

What I can say is that the code compiles and works correctly when using Ubuntu 18 and the version 7.2.1 of the GNU tools. (my current setup)

```
~ $ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Gmlmk1, 2019-04-12:
I noticed that everything works in debug mode, but in release mode the node doen't start. 
Using gpio it turned out that the program is stuck in an infinite loop in here - https://github.com/Lora-net/LoRaMac-node/blob/1cdd9ccec4c9f05b616e7112059be4a9e358c571/src/boards/mcu/stm32/STM32L1xx_HAL_Driver/Src/stm32l1xx_hal_rcc.c#L617 , because function HAL_GetTick() always return '0'.

mluis1, 2019-04-18:
Have you tried to compile the code from develop branch.

I don't remember well now but I guess that a few days after the 4.4.1 release we fixed a similar issue.

Gmlmk1, 2019-04-22:
I download code from master branch. I just added `GpioWrite(&led3,1);` after board initialization function

mluis1, 2019-04-23:
Could you please apply the fix that we provided on commit 44a67d4f5abf675215f308972b1add94c3365485?

I would recommend that you switch your code base to develop branch. We have corrected a lot of issues since last official release and we have also added some improvements to the implementation. 
The develop branch is currently in a state close to be released. We have currently 3 open issues that we would like to fix before the release. (#606, #729, #730)

Gmlmk1, 2019-04-26:
This commit fixed it. But the binary file size has grown.

mluis1, 2019-04-26:
That's the drawback of removing the `-lfto` option.

At the moment we don't know why the code generated with that option is not functional. In plus we didn't had the time to further investigate this.

closed 2019-04-26 by mluis1