Lora-net/LoRaMac-node #719: Does SX1262 Support HFSS?

vansinba, 2019-04-09:
SX1276 do support Hopping mode in Datasheet.But I can't find any information about Frequency hopping spread spectrum in SX1262's Datasheet and code.

mluis1, 2019-04-10:
What do you mean by HFSS?

If it is intra-packet frequency hopping then I guess that you are right the SX1262 doesn't support or document it.
Now if you are talking about frequency hopping per packet then the SX1262 supports it. The US915 region on this project implements a frequency hopping system.

Could you please post your question on the [LoRa Community](https://semtech.force.com/lora) portal.
You will get better support on this kind of questions.

closed 2019-04-18 by mluis1

kokroo, 2024-01-21:
> SX1276 do support Hopping mode in Datasheet.But I can't find any information about Frequency hopping spread spectrum in SX1262's Datasheet and code.

Did you ever find anything like this? I'm also looking for intra-packet hopping on the Sx1262 series.

kokroo, 2024-02-01:
> The US915 region on this project implements a frequency hopping system.
@mluis1 I am unable to find the code related to FHSS for LoRa peer-to-peer on SX1262 for the US915 region in this repository. We have a prototype based on the SX1262 wherein the average time on air is around 5 seconds, and hence we really need frequency hopping to work, to get the device approved by the FCC.

mluis1, 2024-02-01:
As previously stated the SX1262 radio does not support intra-packet frequency hopping.

This project implemented frequency hopping system is the one defined for the LoRaWAN protocol.

In order to operate under FCC rules the LoRaWAN specification does not use SF12BW125 and SF11BW125 data rates as it is impossible to fulfill the 400 ms dwell-time.
The first usable data rate is SF10BW125 however the maximum payload length that can be used is limited to 24 bytes in order to fulfill the 400 ms dwell-time.

Please refer to [LoRaWAN regional parameters](https://resources.lora-alliance.org/technical-specifications/rp002-1-0-4-regional-parameters) specification chapter "2.5 US902-928 MHz ISM Band" for further information.