Lora-net/LoRaMac-node #721: Data frames are not encrypted and uploaded, and frame counts are not increased.

yanzi418, 2019-04-10:
static uint8_t LoRaMacCheckForTxTimeout( void )
{
    if( ( LoRaMacConfirmQueueGetStatusCmn( ) == LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT ) ||
        ( MacCtx.McpsConfirm.Status == LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT ) )
    {
        // Stop transmit cycle due to tx timeout
        MacCtx.MacState &= ~LORAMAC_TX_RUNNING;
        MacCtx.McpsConfirm.NbRetries = MacCtx.AckTimeoutRetriesCounter;
        MacCtx.McpsConfirm.AckReceived = false;
        MacCtx.McpsConfirm.TxTimeOnAir = 0;
        return 0x01;
    }
    return 0x00;
}

 LoRaMacConfirmQueueGetStatusCmn( ) == LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT

This flag bit is not cleared, resulting in no increase in frame count, causing data false alarm to duplicate data frames, resulting in no encrypted data frame reporting.

yanzi418, 2019-04-10:
You can manually set the bit to LORAMAC_EVENT_INFO_STATUS_TX_TIMEOUT, and the data frame will not be encrypted and can not be restored.I think this is a BUG.

yanzi418, 2019-04-10:
develop code lorawan1.03

mluis1, 2019-04-10:
We are currently looking for a suitable solution. Please refer to pull-request #606

closed 2019-04-10 by mluis1