Lora-net/LoRaMac-node #722: LoRA can't transmit more than 64 byte

pointone, 2019-04-10:
I tried to make my project based on the example of ping-pong Lora mode and SX1278. If I send a packet more then 64 bytes, the receiver does not receive it. Smaller packages are accepted normally. BUFFER_SIZE in the main.c file has been increased accordingly. Can master branch implementation be used to send packets larger than 64 bytes (and smaller than 256 bytes)?

trick2011, 2019-04-15:
What SF (spreading factor)/DR (datarate) are you using? You won't be able to transmit more than 51 bytes by LoRaWAN specifications if you are using SF12 (at least in the EU area (868MHz)). There are even LoRaWAN networks which don't even allow bigger messages because a node is then unable to transmit its data when the connection is poor.

pointone, 2019-04-16:
We operate at 434 MHz and use the following settings:

#define TX_OUTPUT_POWER                      20  // dBm
#define LORA_BANDWIDTH   	              0  // 0: 125 kHz
#define LORA_SPREADING_FACTOR            7  // SF7..SF12
#define LORA_CODINGRATE                        1  // 1: 4/5
#define LORA_PREAMBLE_LENGTH              8  // Same for Tx and Rx
#define LORA_FIX_LENGTH_PAYLOAD_ON  false
#define LORA_IQ_INVERSION_ON               false
#define TX_TIMEOUT_VALUE                       3000

Radio.SetTxConfig( MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                   LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                   LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                   true, 0, 0, LORA_IQ_INVERSION_ON, TX_TIMEOUT_VALUE );
    
Radio.SetRxConfig( MODEM_LORA, LORA_BANDWIDTH, LORA_SPREADING_FACTOR,
                   LORA_CODINGRATE, 0, LORA_PREAMBLE_LENGTH,
                   LORA_SYMBOL_TIMEOUT, LORA_FIX_LENGTH_PAYLOAD_ON,
                   0, true, 0, 0, LORA_IQ_INVERSION_ON, true );

fahad80, 2019-04-25:
@pointone The Ping-Pong example has this line:

```#define BUFFER_SIZE                                 64 // Define the payload size here```

closed 2019-06-07 by mluis1