Lora-net/LoRaMac-node #724: RTC_Alarm_IRQn error

selimsagir, 2019-04-12:
I am trying to add LoRamac-node to my GPS and LCD example for using the LoRa communication with b-l072z-lrwan1 kit. But in the "board-rtc.c" file, i am face with "undeclared RTC_Alarm_IRQn" for "HAL_NVIC_SetPriority( RTC_Alarm_IRQn, 4, 0 )". So i changed it to "HAL_NVIC_SetPriority( RTC_IRQn, 4, 0 )" and no error occured. But with that suitable for run the library correctly? And also i have a "rtc.c" file. Will this file be a problem in the following stages?

mluis1, 2019-04-17:
We don't know which branch you are using but, we have searched on master, develop and feature/5.0.0 branches and we don't observe the same as you do.

https://github.com/Lora-net/LoRaMac-node/blob/cd3dc7f8ff25bd91b162f8a627be3d5f7f812218/src/boards/B-L072Z-LRWAN1/rtc-board.c#L195

As you can see the file uses `RTC_IRQn` definition.

Concerning your `rtc.c` file we don't know if there is potential issue as we don't know its contents.

selimsagir, 2019-04-17:
Thanks for your answer. I got file from Semtech web site. So I don't know which version I am using. 
This is my rtc.c file.
```
/* Includes ------------------------------------------------------------------*/
#include "rtc.h"

/* USER CODE BEGIN 0 */

/* USER CODE END 0 */

RTC_HandleTypeDef hrtc;

/* RTC init function */
void MX_RTC_Init(void)
{
  RTC_TimeTypeDef sTime = {0};
  RTC_DateTypeDef sDate = {0};
  RTC_AlarmTypeDef sAlarm = {0};

  /**Initialize RTC Only 
  */
  hrtc.Instance = RTC;
  hrtc.Init.HourFormat = RTC_HOURFORMAT_24;
  hrtc.Init.AsynchPrediv = 31;
  hrtc.Init.SynchPrediv = 1023;
  hrtc.Init.OutPut = RTC_OUTPUT_DISABLE;
  hrtc.Init.OutPutRemap = RTC_OUTPUT_REMAP_NONE;
  hrtc.Init.OutPutPolarity = RTC_OUTPUT_POLARITY_HIGH;
  hrtc.Init.OutPutType = RTC_OUTPUT_TYPE_OPENDRAIN;
  if (HAL_RTC_Init(&hrtc) != HAL_OK)
  {
    Error_Handler();
  }

  /* USER CODE BEGIN Check_RTC_BKUP */
    
  /* USER CODE END Check_RTC_BKUP */

  /**Initialize RTC and set the Time and Date 
  */
  sTime.Hours = 0;
  sTime.Minutes = 0;
  sTime.Seconds = 0;
  sTime.DayLightSaving = RTC_DAYLIGHTSAVING_NONE;
  sTime.StoreOperation = RTC_STOREOPERATION_RESET;
  if (HAL_RTC_SetTime(&hrtc, &sTime, RTC_FORMAT_BIN) != HAL_OK)
  {
    Error_Handler();
  }
  sDate.WeekDay = RTC_WEEKDAY_MONDAY;
  sDate.Month = RTC_MONTH_JANUARY;
  sDate.Date = 1;
  sDate.Year = 0;

  if (HAL_RTC_SetDate(&hrtc, &sDate, RTC_FORMAT_BIN) != HAL_OK)
  {
    Error_Handler();
  }
  /**Enable the Alarm A 
  */
  sAlarm.AlarmTime.Hours = 0;
  sAlarm.AlarmTime.Minutes = 0;
  sAlarm.AlarmTime.Seconds = 0;
  sAlarm.AlarmTime.SubSeconds = 0;
  sAlarm.AlarmTime.DayLightSaving = RTC_DAYLIGHTSAVING_NONE;
  sAlarm.AlarmTime.StoreOperation = RTC_STOREOPERATION_RESET;
  sAlarm.AlarmMask = RTC_ALARMMASK_NONE;
  sAlarm.AlarmSubSecondMask = RTC_ALARMSUBSECONDMASK_NONE;
  sAlarm.AlarmDateWeekDaySel = RTC_ALARMDATEWEEKDAYSEL_DATE;
  sAlarm.AlarmDateWeekDay = 1;
  sAlarm.Alarm = RTC_ALARM_A;
  if (HAL_RTC_SetAlarm_IT(&hrtc, &sAlarm, RTC_FORMAT_BIN) != HAL_OK)
  {
    Error_Handler();
  }

}

void HAL_RTC_MspInit(RTC_HandleTypeDef* rtcHandle)
{

  if(rtcHandle->Instance==RTC)
  {
  /* USER CODE BEGIN RTC_MspInit 0 */

  /* USER CODE END RTC_MspInit 0 */
    /* RTC clock enable */
    __HAL_RCC_RTC_ENABLE();

    /* RTC interrupt Init */
    HAL_NVIC_SetPriority(RTC_IRQn, 0, 0);
    HAL_NVIC_EnableIRQ(RTC_IRQn);
  /* USER CODE BEGIN RTC_MspInit 1 */

  /* USER CODE END RTC_MspInit 1 */
  }
}

void HAL_RTC_MspDeInit(RTC_HandleTypeDef* rtcHandle)
{

  if(rtcHandle->Instance==RTC)
  {
  /* USER CODE BEGIN RTC_MspDeInit 0 */

  /* USER CODE END RTC_MspDeInit 0 */
    /* Peripheral clock disable */
    __HAL_RCC_RTC_DISABLE();

    /* RTC interrupt Deinit */
    HAL_NVIC_DisableIRQ(RTC_IRQn);
  /* USER CODE BEGIN RTC_MspDeInit 1 */

  /* USER CODE END RTC_MspDeInit 1 */
  }
} 

/* USER CODE BEGIN 1 */

/* USER CODE END 1 */

/************************ (C) COPYRIGHT STMicroelectronics *****END OF FILE****/
```

mluis1, 2019-04-17:
Yes, there will be a potential as the `MX_RTC_Init` in your file is doing the same thing as the `RtcInit` on this project. ( the other functions on your file have their equivalent on the `rtc-board.c` file)

In this project the `rtc-board.c` file replaces the one you have.

selimsagir, 2019-04-17:
The problem is that, I am trying to make a project including GPS module and LCD screen besides LoRa communication with b-l072z-lrwan1 kit. I was using library from ST but It was so complex for add GPS and LCD. So I have made a project with CubeMX and have done GPS and LCD successfully. Then I have found this library to add LoRa communication but I could not make it success. What would be your advice for this project. 
[https://github.com/selimsagir/STM32_LoRa/tree/master/GPS_Demo/Src](url)

mluis1, 2019-04-17:
This project as is, already provides driver examples for I2C and UART based GPS modules.
UART - https://github.com/Lora-net/LoRaMac-node/blob/develop/src/boards/NAMote72/gps-board.c
I2C - https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/MoteII/gps-board.c

There is also an example of a LCD display driver on master branch for the MoteII platform (this platform support has ended in develop branch)
https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/MoteII/display-board.c
https://github.com/Lora-net/LoRaMac-node/blob/master/src/apps/LoRaMac/classA/MoteII/screen.c

I think that the easiest would be to keep this project structure.
If you want to keep your structure then you need to update the files generated by CubeMX with the contents of the equivalent files on this project.

Another possibility would be to use the `i-cube-lrwan` software done by ST which is based on this project. https://www.st.com/en/embedded-software/i-cube-lrwan.html

I close this issue as there is not much more things we can do for you on this subject.

closed 2019-04-17 by mluis1