Lora-net/LoRaMac-node #725: Max number of channels defined in AS923 vs CFList

ellecto, 2019-04-15:
The AS923_NUMB_CHANNELS_CF_LIST is defined to be 5 in RegionAS923.h and there are these lines in RegionAS923.c:

```
    // Last byte is RFU, don't take it into account
    for( uint8_t i = 0, chanIdx = AS923_NUMB_DEFAULT_CHANNELS; chanIdx < AS923_MAX_NB_CHANNELS; i+=3, chanIdx++ )
    {
        if( chanIdx < ( AS923_NUMB_CHANNELS_CF_LIST + AS923_NUMB_DEFAULT_CHANNELS ) )
        {
            // Channel frequency
```

The above if does not allow more than 5 extra channels to be defined by the CFList.

Looking into Loriot and TTN, most AS923 plans defined 8 channels. So, probably 6 extra channels will be sent to the device, the last one will be ignored.

This will result in the device ignoring ADR change request since there is a mismatch between the number of channels defined in the network server and the device.

I wonder how with this limitations most (if not all) the AS923 plans defined 8 channels? 
Is the last frequency required to be entered manually? What am I missing here?

ellecto, 2019-04-15:
Also, just realised if the 6th channel is added manually using `RegionAS923ChannelAdd()`, it'll be ignored automatically anyway as the channel ID exceeds the allowed 5.

In the regional spec there is a provision for 16 channels for this plan....

mluis1, 2019-04-18:
The regional specification [LoRaWAN 1.0.3 Regional Parameters](https://lora-alliance.org/sites/default/files/2018-07/lorawan_regional_parameters_v1.0.3reva_0.pdf) says the following:

[image]

Which means that the end-device must allocate space to store at least 16 channels.

If you look at chapter 2.8.4 of the same specification you will notice that the CFList can only transport 5 channels over the air.

[image]

If more channels are required then the network server must setup them using the `NewChannelReq/Ans` MAC command.

closed 2019-04-18 by mluis1