Lora-net/LoRaMac-node #731: Changing frequency plan at runtime (sort of)

jeremyherbert, 2019-04-18:
Is there any mechanism to switch frequency plans at "runtime"? I put runtime in quotes, because I mean before the stack has been initialised (obviously once the stack has been initialised, it wouldn't really make sense to change plans). AKA is it possible to compile in multiple frequency plans, and choose which to use at the point of initialisation? In particular, there is an issue in Australia where both AU915 and AS923 are valid and acceptable frequency plans, and there are several networks using one or the other (but not both).

mluis1, 2019-04-18:
Yes, there is.

When executing `cmake` command you can enable more than one frequency plan (all plans enabled is also possible). Then in the main application you may initialize the stack with the wanted frequency plan.

There is also the possibility to dynamically change the plan by re-initializing the stack with another plan. This will act as if the end-device was reset.

In your case you can execute `cmake` as follows:
```bash
$ cmake -DCMAKE_BUILD_TYPE=Debug \
        -DTOOLCHAIN_PREFIX="/path/to/toolchain" \
        -DCMAKE_TOOLCHAIN_FILE="../cmake/toolchain-arm-none-eabi.cmake" \
        -DAPPLICATION="LoRaMac" \
        -DCLASS="classA" \
        -DCLASSB_ENABLED="OFF" \
        -DACTIVE_REGION="LORAMAC_REGION_AU915" \
        -DREGION_EU868="OFF" \
        -DREGION_US915="OFF" \
        -DREGION_CN779="OFF" \
        -DREGION_EU433="OFF" \
        -DREGION_AU915="ON" \
        -DREGION_AS923="ON" \
        -DREGION_CN470="OFF" \
        -DREGION_KR920="OFF" \
        -DREGION_IN865="OFF" \
        -DREGION_RU864="OFF" \
        -DBOARD="NucleoL073" \
        -DMBED_RADIO_SHIELD="SX1262DVK1CAS"  ..
```

jeremyherbert, 2019-04-18:
Great! Thank you. I will give that a go.

closed 2019-04-18 by jeremyherbert