Lora-net/LoRaMac-node #732: Question, Is ClassB work?

PellegrinoA, 2019-04-18:
Somebody did work the Class B? or the code is in develop?
Because I see that the code has a lot of problem and in my case the beacon isn't received but when I use the solution #552 only receive the first beacon.

mluis1, 2019-04-23:
Yes, the ClassB example is generally working. Although there are 2 open issues that are currently under investigation #729 and #730.

We are validating the current implementation of ClassB using Actility and loraserver.io network servers.

PellegrinoA, 2019-04-23:
Ok, Because I use AU915 plan and the frequency beacon is bad calculated. The node never catch the beacons but if I use the frequency fix then the node catch only the first beacon. The node change class A to B but never catch other beacon.  
I'm using loraserver.io too. I supouse that the problem is in the nodo. 

This class is a challenge!!!

mluis1, 2019-04-24:
I'm currently validating the fix for issues #729 and #730 and I guess that it will also fix the issue that you are observing on the beacon hopping.

It would be nice if you could confirm that the fix solves your issue once I have pushed it.

mluis1, 2019-04-25:
I think that the commit 6bead5f0292b171ccc307c5a892fc241b2635373 should fix the beacon hopping issues that you are observing.

Could you please check the ClassB operation using the updated develop branch?

Thanks in advance.

closed 2019-06-07 by mluis1