Lora-net/LoRaMac-node #733: What parameters need to be saved to restore join context on reset

NicoHell1, 2019-04-19:
Hi,

I was wondering what needed to be saved in order to restore the stack after a system reset. This could help me save energy on reset by not joining when not needed.

I already search on this repo for a similar issue, I found #213 interesting, but I need if possible an exhaustive list of what need to be saved in EEPROM, for now I saved:
* uplink & downlink counters
* netID
* appSKey
* nwkSKey
* devAddr
* all LoRaMacParams.

It does not seem to work right now so I guess something is missing (CFList, ChannelsParams, something else ?). I work with last release on master so V4.4.1 with LoRaWAN specification 1.02

If you need any further detail I will try to provide it, thanks

mluis1, 2019-04-23:
Please take a look at the [NvmCtxMgmt.h](https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/LoRaMac/common/NvmCtxMgmt.h) and [NvmCtxMgmt.c](https://github.com/Lora-net/LoRaMac-node/blob/develop/src/apps/LoRaMac/common/NvmCtxMgmt.c) files.

The `MAX_PERSISTENT_CTX_MGMT_ENABLED` pre-processing definition allows to choose between the minimum and full context store/restore.

NicoHell1, 2019-04-23:
Hello @mluis1, thank you for your answer, but as I said, I am working with the latest release on master branch, which does not provide these files, I tried to look at them anyway but I did not fully understand what is saved

fahad80, 2019-04-25:
@NicoHell1 I did something similar with the old branch. And I saved the things you mentioned (except 'all LoRaMacParams') and it worked for me.

NicoHell1, 2019-04-25:
@fahad80 Ah, that is a bit weird then, I thought that it was indeed enough to get what I needed, but when my device tried to send an uplinkthe gateway did not seem to receive any, I will have to verify that again

mluis1, 2019-04-25:
Sorry I referenced the wrong issue. I wanted to reference issue #732

fahad80, 2019-04-26:
@NicoHell1 Did you mean your network server didn't receive the uplink or the gateway didn't receive the uplink. 

Because, 'usually', the gateway should receive the packet. If something is wrong (for example keys, counters etc.), NS will reject it.

But if GW doesn't receive it, I would say, check the channel params.

closed 2019-04-29 by NicoHell1