Lora-net/LoRaMac-node #735: How to increase the TX power up to +20dbm

sagar-patel-sls, 2019-04-25:
Hii
how can we increase TX power up to +20dbm for EU433?

```
board: sx1276
ISM band: EU433
MBED_RADIO_SHIELD: SX1276MB1MAS
```

Thank you

mluis1, 2019-04-25:
The LoRaWAN regional parameters specification doesn't allow this. Please refer to the specification.

[image]

Maximum allowed EIRP is 12.5 dBm

closed 2019-04-25 by mluis1

sagar-patel-sls, 2019-04-25:
Thank you @mluis1

sagar-patel-sls, 2019-04-25:
@mluis1 

we are getting only 100m range. what we can do for maximum range?

mluis1, 2019-04-25:
It may depend on your PCB design, PCB antenna matching, used antenna, etc...

As this is not related to this software project it would be nice if you post this kind of questions on Semtech forums (please select the right topic). https://lora-developers.semtech.com/knowledge-base/forum/