Lora-net/LoRaMac-node #736: Support for ESP32 and Arduino ide

divya0056, 2019-04-26:
Is it possible to add support for Arduino IDE and MCU support for esp32. Thanks

deandob, 2019-04-28:
+1. 
I'm about to undertake a port of point to point SX1262 to the ESP32 - if anyone else has done or doing this, please share to give me a headstart,

mluis1, 2019-06-07:
As we you may know we are a small team and it is impossible for us to support every MCU or development environment existent in the world (there are too many).

We think that the project is designed in such a way that it should be an easy task for a developer to adapt it to a new development environment or to target a different processor.

You may have better answers to your request by posting it on [TTN](https://www.thethingsnetwork.org/forum/) forum or on [Semtech developers portal forum](https://lora-developers.semtech.com/knowledge-base/forum/).

closed 2019-06-07 by mluis1