Lora-net/LoRaMac-node #737: RxTimeoutSyncWord eq 0

Wielebny666, 2019-05-07:
Hi,
I use MODEM_FSK.
In:
https://github.com/Lora-net/LoRaMac-node/blob/0100a2188726c814d2ab7b974b9e940e449c1504/src/apps/ping-pong/SAML21/main.c#L200
symbTimeout is 0

https://github.com/Lora-net/LoRaMac-node/blob/0100a2188726c814d2ab7b974b9e940e449c1504/src/radio/sx1276/sx1276.c#L418
SX1276.Settings.Fsk.RxSingleTimeout.RxSingleTimeout also 0

https://github.com/Lora-net/LoRaMac-node/blob/0100a2188726c814d2ab7b974b9e940e449c1504/src/radio/sx1276/sx1276.c#L1021

I cant set RxTimeoutSyncWord timer to 0.

sidpat, 2019-05-19:
According to the implementation of the `TimerSetValue` function, a minimum timer value is set in the rtc-board.c file which is 3 sub-second ticks. This is done so that the RTC time doesnt go ahead of the alarm time during the process of setting the alarm.

mluis1, 2019-06-07:
Can we close the issue?

mluis1, 2019-06-13:
No feedback. We assume that we can close the issue.

closed 2019-06-13 by mluis1