Lora-net/LoRaMac-node #740: [nRF52832 + RFM95] node

davidgmoscoso, 2019-05-10:
Hi everyone, i'm currently developing a [nRF52832 + RFM95] node. I want to be able to send messages to TTN in US915 frequency plan using ABP, no softdevice for the moment, as simple as possible. I found this code from @RAKWireless [https://github.com/RAKWireless/RAK813_LoRaNode](url) that was developed for one of their boards (nRF52832 + sx1276). I simply ported and edited it on Segger Studio to adapt to my own board, the code compile succesfully but i don't get any message on the TTN console.
Do you have any experence with nrf52? Maybe some things i've missed
Any advice will be helpful, thanks

mluis1, 2019-06-07:
Unfortunately we don't have much experience on those platforms.

You may have better answers to your question by posting it on [TTN](https://www.thethingsnetwork.org/forum/) forum or on [Semtech developers portal forum](https://lora-developers.semtech.com/knowledge-base/forum/).

closed 2019-06-07 by mluis1

sridhar2rok, 2021-07-06:
@davidgmoscoso, are you successful in building the Node with nRF52832 + RFM95?  I am finding issues with the RAK813 code.