Lora-net/LoRaMac-node #741: set of the paConfig register in the step of enable 20dbm

caoenjie, 2019-05-13:
Hi
    I would like to enable the PA_Boost to have +20dBm in the radio。
    **paConfig = 0xFF or 0xCF ?**
    _Maxpower_（4-6） = 100（default）or 111（max）?    
    In addition ，I found `paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK ) | 0x70; ` deleted in SetRfTxPower function。Can you tell me the reason？
    thanks！

mluis1, 2019-05-14:
Sorry, but we don't fully understand your question.

Where in the code provided here the line you referenced is deleted.

To set the RF output power to 20 dBm on SX1276 radio you just need, in the code provided on this repository, to call  `SX1276SetRfTxPower( 20 );`

Please refer to the different sx1276xxxx-board.c files.
Example 
https://github.com/Lora-net/LoRaMac-node/blob/68dfa1270b9e1073e8ae375f6ee338fd35d85af5/src/boards/NucleoL073/sx1276mb1las-board.c#L176

```c
void SX1276SetRfTxPower( int8_t power )
{
    uint8_t paConfig = 0;
    uint8_t paDac = 0;

    paConfig = SX1276Read( REG_PACONFIG );
    paDac = SX1276Read( REG_PADAC );

    paConfig = ( paConfig & RF_PACONFIG_PASELECT_MASK ) | SX1276GetPaSelect( power );

    if( ( paConfig & RF_PACONFIG_PASELECT_PABOOST ) == RF_PACONFIG_PASELECT_PABOOST )
    {
        if( power > 17 )
        {
            paDac = ( paDac & RF_PADAC_20DBM_MASK ) | RF_PADAC_20DBM_ON;
        }
        else
        {
            paDac = ( paDac & RF_PADAC_20DBM_MASK ) | RF_PADAC_20DBM_OFF;
        }
        if( ( paDac & RF_PADAC_20DBM_ON ) == RF_PADAC_20DBM_ON )
        {
            if( power < 5 )
            {
                power = 5;
            }
            if( power > 20 )
            {
                power = 20;
            }
            paConfig = ( paConfig & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( uint8_t )( ( uint16_t )( power - 5 ) & 0x0F );
        }
        else
        {
            if( power < 2 )
            {
                power = 2;
            }
            if( power > 17 )
            {
                power = 17;
            }
            paConfig = ( paConfig & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( uint8_t )( ( uint16_t )( power - 2 ) & 0x0F );
        }
    }
    else
    {
        if( power > 0 )
        {
            if( power > 15 )
            {
                power = 15;
            }
            paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( 7 << 4 ) | ( power );
        }
        else
        {
            if( power < -4 )
            {
                power = -4;
            }
            paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( 0 << 4 ) | ( power + 4 );
        }
    }
    SX1276Write( REG_PACONFIG, paConfig );
    SX1276Write( REG_PADAC, paDac );
}
```

caoenjie, 2019-05-15:
Fisrt of all,thank you for your  reply!
I'm sorry that I did not describe this question clearly enough. In fact, I'm in trouble when setting the 20dBm transmit power. I can write to **0xFF** in the **paConfig register** to enable 20dBm, but **0xCF** does not work.  I compared the `SX1276SetRfTxPower( )` functions of the old and new versions.
I found `paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK ) | 0x70; `deleted in new version。
If I want to set 20dBm, the value of paconfig in the old version is **0xFF**(maybe` paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK ) | 0x70;` is work )，but in the new version it's **0xCF**。So, I check the datasheet and I have two question here：
1，In paConfig register ,_Maxpower_（4-6)  = 100 or 111 ? If I set 20dBm, does the different settings affect this?
2，Why did you delete  `paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK ) | 0x70;` in current version? 
By the way, I'm a novice ,please understand if the description is unclear. :)
thanks!

mluis1, 2019-05-15:
I'm sorry but it is still not very clear what you would like to achieve.
Why not just use the function as it is proposed. It will setup the radio to transmit at 20 dBm.

Again could you please point out where in the code the `paConfig = ( paConfig & RF_PACONFIG_MAX_POWER_MASK ) | 0x70;` has been deleted.

The current develop branch code as shown on my previous answer has the same line (at least in functionality).
https://github.com/Lora-net/LoRaMac-node/blob/68dfa1270b9e1073e8ae375f6ee338fd35d85af5/src/boards/NucleoL073/sx1276mb1las-board.c#L229

`( 7 << 4 ) = 0x70`

Please also note that the line you point out is only used when the radio output is `RFO` where the maximum output power is 15 dBm.

When setting the radio to operate in 20 dBm there are some other settings that need to be changed and the formula used for the register setting changes (`paDac` register).

Have you changed the function `SX1276GetPaSelect` to return `RF_PACONFIG_PASELECT_PABOOST` value as you would like to have the radio to output 20 dBm?

caoenjie, 2019-05-20:
The function I am referring to is in **LoRaMac-node/src/boards/B-L072Z-LRWAN1/sx1276-board.c**，When power is _20dBm_, function _SX1276GetPaSelect_ returns **RF_PACONFIG_PASELECT_PABOOST**。
so,`if( paConfig & RF_PACONFIG_PASELECT_PABOOST ) == RF_PACONFIG_PASELECT_PABOOST` Should be true。Then，`paConfig = ( paConfig & RF_PACONFIG_OUTPUTPOWER_MASK ) | ( uint8_t )( ( uint16_t )( power - 5 ) & 0x0F );` Should be executed。
 paConfig = **0xCF** is the final result。
But in my program, only when paConfg=0xFF will be enable 20dBm。The difference between 0xCF and 0xFF is in the 4-6。In datasheet，the meaning of 4-6 is _Maxpower_。
So，do you think _Maxpower_ should be 100 or 111 ?  If I set 20dBm, does the different settings affect this?
Thanks ！

mluis1, 2019-05-20:
If you look at formulas on SX1276 you will notice that `Pmax` (bits 6-4) are not used when PA_BOOST is used.

The datasheet provided formulas must be applied as is.
[image]

Again if you use the provided function as is, you will get 20 dBm as output power. You must understand that the 20 dBm operation is a special mode which requires in plus of the `RegPaConfig` register the setup of `RegPaDac` register.

If the `RegPaDac` register is not changed, `0x84`, the `Pmax` will be 17 dBm.
If the `RegPaDac` register is changed to `0x87` the `Pmax` will be 20 dBm.

[image]

As this question is more related to the radio chip usage I would recommend that you post your question on the [lora-developers](https://lora-developers.semtech.com/knowledge-base/forum/) forum.

caoenjie, 2019-05-21:
I understand，I appreciate your support。

closed 2019-05-21 by caoenjie