Lora-net/LoRaMac-node #742: US915 Limit channels

utkarshshah007, 2019-05-14:
Hey folks!

I'm new here and I'm trying to improve the join time for my node. I'm using an 8 channel gateway & TTN as my network server.

I'd like to limit the channels that the device tries to join on based on this: https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html#us902-928

I tried to do this like so (as I've seen in similar issues):
```
// Limiting to just 2nd subband of 8 channels
                static uint16_t GatewayChannelsMask[6] = {0xFF, 0x00, 0x00, 0x00, 0x00, 0x00};
                mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
                mibReq.Param.ChannelsDefaultMask = GatewayChannelsMask;
                LoRaMacMibSetRequestConfirm( &mibReq );

                mibReq.Type = MIB_CHANNELS_MASK;
                mibReq.Param.ChannelsMask = GatewayChannelsMask;
                LoRaMacMibSetRequestConfirm( &mibReq );
```
but it did not seem to work.

I don't quite understand how the ChannelsMask works, and I was wondering if someone can help me understand what I'm doing wrong.

mluis1, 2019-05-14:
You are setting the channel mask for what they call 1st sub-band. That's why it isn't working.

Please note that you should only use this trick for debug purposes. If you keep this your end-device cannot be certified, as for LoRaWAN compliance all 64 channels must be used.
Please also note that the Join procedure algorithm has been updated in order to speed up the join procedure. Please refer to the latest regional parameters specification. 

[image]

The correct way to do what you would like is shown below:

```c
#if defined( REGION_US915 )
                // Enabling 2nd block of 8 channels (8-15) + channel 65
                uint16_t channelMask[] = { 0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000};
                mibReq.Type = MIB_CHANNELS_MASK;
                mibReq.Param.ChannelsMask = channelMask;
                LoRaMacMibSetRequestConfirm( &mibReq );
                mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
                mibReq.Param.ChannelsDefaultMask = channelMask;
                LoRaMacMibSetRequestConfirm( &mibReq );
#endif
```

utkarshshah007, 2019-05-16:
Thanks @mluis1! This helped a lot - join is much faster, and RSSI values are improved as well.

Which version of this library added the faster join procedure algorithm for all 64 channels? 
I am currently using the latest version available at mbed: https://os.mbed.com/teams/Semtech/code/LoRaWAN-demo-72/
However, that appears to be several years old.

closed 2019-05-16 by utkarshshah007

MedadRufus, 2020-10-05:
```
#if defined( REGION_US915 )
                // Enabling 2nd block of 8 channels (8-15) + channel 65
                uint16_t channelMask[] = { 0xFF00, 0x0000, 0x0000, 0x0000, 0x0002, 0x0000};
                mibReq.Type = MIB_CHANNELS_MASK;
                mibReq.Param.ChannelsMask = channelMask;
                LoRaMacMibSetRequestConfirm( &mibReq );
                mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
                mibReq.Param.ChannelsDefaultMask = channelMask;
                LoRaMacMibSetRequestConfirm( &mibReq );
#endif
```
Can I confirm, do we still need to use this channel mask when using The Things Network on US915 frequencies? Is this the case for loramac-node  v4.4.2-v4.4.4? Will it work if we use Activation By Personalisation(ABP) instead of Over the Air Activation(OTAA)? In addition, is this channel mask necessary for regions AU915 and AS923 when using The Things Network? 

Thanks
Medad