Lora-net/LoRaMac-node #743: RtcGetCalendarValue() - RTC readout synchronisation mechanism bug

Oanerer, 2019-05-16:
We’ve encountered an issue in the ST stack (and StackForce stack) that caused our nodes to stay in STOP mode for 17.5 days. After weeks of searching, the problem can be traced back to the RTC readout synchronisation mechanism used in RTCGetCalendarValue():

```// Get Time and Date
HAL_RTC_GetTime( &RtcHandle, time, RTC_FORMAT_BIN );

// Make sure it is correct due to asynchronus nature of RTC
do
{
    firstRead = time->SubSeconds;
    HAL_RTC_GetDate( &RtcHandle, date, RTC_FORMAT_BIN );
    HAL_RTC_GetTime( &RtcHandle, time, RTC_FORMAT_BIN );
}while( firstRead != time->SubSeconds );
```
The synchronisation mechanism is based on the `time->SubSeconds` variable. When this variable has not changed between two consecutive read actions, the contents read of the RTC->TR (time) and RTC->DR (date) registers are considered valid. This assumption is false however, as we have experienced first hand...

In the HAL_RTC_GetTime() function, RTC->SSR is read before RTC->TR. This means that in between reading RTC->SSR and RTC->TR, the RTC may have actually advanced one tick. In 1023 out of 1024 times, this is not a problem. Consider however the following scenario:

Before entering the do->while loop, the first call to HAL_RTC_GetTime() yields a seconds value of 10 and a SubSeconds value of 1023. `firstRead` then contains the value 1023. All fine. 
Then we enter HAL_RTC_GetTime() for a second time. Now, in between the reads of RTC->SSR and RTC->TR the RTC advances one tick. This means that `time->SubSeconds` will still contain the value 1023 since RTC->SSR was already read. However, the seconds value has advanced to 11, which we retrieve after reading RTC->TR. The subseconds value at that time is actually 0, but we do not know about this since we've read that register earlier!!

When returning from HAL_RTC_GetTime() we compare `firstRead` with `time->SubSeconds` and find that both values are identical. RtcGetCalendarValue() therefore considers the current time to be 11 seconds, 1023 ticks, while the actual time is 11 seconds 0 ticks!

The real issue occurs when RtcGetTimerElapsedTime() comes into play. At line 443 this function subtracts the current calendar value from `RtcTimerContext.Time` - a calendar value set when RtcSetTimerContext() is called, for example at TimerStart() line 126. When the `RtcTimerContext.Time` variable contains the incorrectly determined calendar value of 1023 ticks in the future, the subtraction in RtcGetTimerElapsedTime() yields a value close to **-1023**. Since the function returns a uint32_t, this value is actually treated as `2^32 - 1023`, or 4294966273. TimerSetTimeout() at line 373 then adds the return value of 'RtcGetTimerElapsedTime() + minTicks' and sets the result to the timestamp of the requested timer. At line 375, this timestamp is then used to set the RTC Alarm through RtcSetAlarm().

From there on, in the function RtcStartAlarm() the incorrect timestamp is added to the current time, etc etc, which results at line 393 in an `rtcAlarmDays` value of 48.5 days + the time since powerup when the issue occurred. The modulo operation from lines 393 to 406 truncate this value to 48.5 days MINUS the amount of days in that month, which is initially January containing 31 days. Hence, our RTC alarm was always set 17.5 days in the future, which caused our node not to wake up from STOP mode for that period, unless explicitly woken up from another source.

From what I can see this issue is present across all rtc-board.c files. The easiest way to solve this is probably to replace the `firstRead` variable of type uint32_t containing only the subSeconds value with a `firstRead` variable of type RTC_TimeTypeDef containing the combined contents of the RTC->TR and RTC->SSR registers. You can then do a memcmp(), or something similar (if you want to stay away from string.h) of `firstRead` and `time` at line 497. From what I can see, that should provide enough assurances that the RTC values are read correctly. 

Let's hope this saves other people some headaches...

Oanerer, 2019-05-16:
Quick program to reproduce this:

```
void HW_RTC_TestLoop( void )
{
	RTC_TimeTypeDef RTC_TimeStruct;
	RTC_DateTypeDef RTC_DateStruct;

	uint32_t next;
	uint32_t prev = HW_RTC_GetCalendarValue(&RTC_DateStruct, &RTC_TimeStruct );
	uint32_t diff;

	while(1)
	{
		next = HW_RTC_GetCalendarValue(&RTC_DateStruct, &RTC_TimeStruct );
		if( next < prev )
		{
			diff = next - prev;
			while(1)
			{
				PRINTF("-------------------------------\r\n");
				PRINTF("Difference unsigned: %u\r\n", diff);
				PRINTF("Difference signed: %d\r\n", (int32_t)diff);

				HAL_Delay(1000);
			}
		}

		prev = next;
	}
}
```

sidpat, 2019-05-16:
Will this be solved by

 `HAL_RTCEx_EnableBypassShadow(&RtcHandle)  // in rtc init function`
```
HAL_RTC_GetTime( &RtcHandle, time, RTC_FORMAT_BIN ); // in RtcGetCalendarValue() function
HAL_RTC_GetDate( &RtcHandle, date, RTC_FORMAT_BIN );
```
?

RTC shadow registers should ensure data integrity while reading.

YoannBo, 2019-05-16:
Hi,

I think we can solve by bracing date&time reading within 2 subsecond register read.
This way we are sure it will be ok.

Using some low level APIs  in #include "stm32l0xx_ll_rtc.h"
it could be:

/*!
 * @brief get current time from calendar in ticks
 * @param pointer to RTC_DateStruct
 * @param pointer to RTC_TimeStruct
 * @retval time in ticks
 */
static uint32_t RTC_GetCalendarValue( RTC_DateTypeDef* RTC_DateStruct, RTC_TimeTypeDef* RTC_TimeStruct )
{
  uint32_t calendarValue = 0;
  uint32_t correction;
#if 1 /*with ll APIs*/
  uint32_t ssr;
  uint32_t tr;
  uint32_t dr;
    do
  {
    ssr = RTC->SSR;
    tr = RTC->TR;
    dr = RTC->DR;
  } while (ssr != RTC->SSR);
  
  RTC_TimeStruct->SubSeconds = ssr;
  RTC_TimeStruct->Hours =   __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_HOUR(tr) );
  RTC_TimeStruct->Minutes = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_MINUTE(tr) );
  RTC_TimeStruct->Seconds = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_SECOND(tr) );
  RTC_DateStruct->WeekDay =   __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_WEEKDAY(dr) );
  RTC_DateStruct->Date =   __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_DAY(dr) );
  RTC_DateStruct->Month =    __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_MONTH(dr) );
  RTC_DateStruct->Year =    __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_YEAR(dr) );
  
#else  
  uint32_t first_read;
  /* Get Time and Date*/
  HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
 

   /* make sure it is correct due to asynchronus nature of RTC*/
  do {
    first_read = RTC_TimeStruct->SubSeconds;
    HAL_RTC_GetDate( &RtcHandle, RTC_DateStruct, RTC_FORMAT_BIN );
    HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
    
  } while (first_read != RTC_TimeStruct->SubSeconds);
#endif 
  /* calculate amount of elapsed days since 01/01/2000 */
  calendarValue= DIVC( (DAYS_IN_YEAR*3 + DAYS_IN_LEAP_YEAR)* RTC_DateStruct->Year , 4);

  correction = ( (RTC_DateStruct->Year % 4) == 0 ) ? DAYS_IN_MONTH_CORRECTION_LEAP : DAYS_IN_MONTH_CORRECTION_NORM ;
 
  calendarValue +=( DIVC( (RTC_DateStruct->Month-1)*(30+31) ,2 ) - (((correction>> ((RTC_DateStruct->Month-1)*2) )&0x3)));

  calendarValue += (RTC_DateStruct->Date -1);
  
  /* convert from days to seconds */
  calendarValue *= SECONDS_IN_1DAY; 

  calendarValue += ( ( uint32_t )RTC_TimeStruct->Seconds + 
                     ( ( uint32_t )RTC_TimeStruct->Minutes * SECONDS_IN_1MINUTE ) +
                     ( ( uint32_t )RTC_TimeStruct->Hours * SECONDS_IN_1HOUR ) ) ;

  calendarValue = (calendarValue<<RTC_N_PREDIV_S) + ( RTC_PREDIV_S - RTC_TimeStruct->SubSeconds);

  return( calendarValue );
}

YoannBo, 2019-05-16:
Hi,
I think we can solve by bracing date&time reading within 2 subsecond register read.
This way we are sure it will be ok.
Using some low level APIs in #include "stm32l0xx_ll_rtc.h"
it could be:
/*!
@brief get current time from calendar in ticks
@param pointer to RTC_DateStruct
@param pointer to RTC_TimeStruct
@RetVal time in ticks
/
static uint32_t RTC_GetCalendarValue( RTC_DateTypeDef RTC_DateStruct, RTC_TimeTypeDef* RTC_TimeStruct )
{
uint32_t calendarValue = 0;
uint32_t correction;
#if 1 /* with ll APIs */
uint32_t ssr;
uint32_t tr;
uint32_t dr;
do
{
ssr = RTC->SSR;
tr = RTC->TR;
dr = RTC->DR;
} while (ssr != RTC->SSR);
RTC_TimeStruct->SubSeconds = ssr;
RTC_TimeStruct->Hours = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_HOUR(tr) );
RTC_TimeStruct->Minutes = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_MINUTE(tr) );
RTC_TimeStruct->Seconds = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_SECOND(tr) );
RTC_DateStruct->WeekDay = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_WEEKDAY(dr) );
RTC_DateStruct->Date = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_DAY(dr) );
RTC_DateStruct->Month = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_MONTH(dr) );
RTC_DateStruct->Year = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_YEAR(dr) );
#else /*current way with issue described*/
uint32_t first_read;
/* Get Time and Date*/
HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
/* make sure it is correct due to asynchronus nature of RTC*/
do {
first_read = RTC_TimeStruct->SubSeconds;
HAL_RTC_GetDate( &RtcHandle, RTC_DateStruct, RTC_FORMAT_BIN );
HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
} while (first_read != RTC_TimeStruct->SubSeconds);
#endif
/* calculate amount of elapsed days since 01/01/2000 /
calendarValue= DIVC( (DAYS_IN_YEAR3 + DAYS_IN_LEAP_YEAR)* RTC_DateStruct->Year , 4);
correction = ( (RTC_DateStruct->Year % 4) == 0 ) ? DAYS_IN_MONTH_CORRECTION_LEAP : DAYS_IN_MONTH_CORRECTION_NORM ;
calendarValue +=( DIVC( (RTC_DateStruct->Month-1)*(30+31) ,2 ) - (((correction>> ((RTC_DateStruct->Month-1)*2) )&0x3)));
calendarValue += (RTC_DateStruct->Date -1);
/* convert from days to seconds */
calendarValue *= SECONDS_IN_1DAY;
calendarValue += ( ( uint32_t )RTC_TimeStruct->Seconds +
( ( uint32_t )RTC_TimeStruct->Minutes * SECONDS_IN_1MINUTE ) +
( ( uint32_t )RTC_TimeStruct->Hours * SECONDS_IN_1HOUR ) ) ;
calendarValue = (calendarValue<<RTC_N_PREDIV_S) + ( RTC_PREDIV_S - RTC_TimeStruct->SubSeconds);
return( calendarValue );
}

YoannBo, 2019-05-16:
Hi,
I think we can solve by bracing date&time reading within 2 subsecond register read.
This way we are sure it will be ok.
Using some low level APIs in #include "stm32l0xx_ll_rtc.h"
it could be:
/*!
@brief get current time from calendar in ticks
@param pointer to RTC_DateStruct
@param pointer to RTC_TimeStruct
@RetVal time in ticks
/
static uint32_t RTC_GetCalendarValue( RTC_DateTypeDef RTC_DateStruct, RTC_TimeTypeDef* RTC_TimeStruct )
{
uint32_t calendarValue = 0;
uint32_t correction;

#if 1  /*with ll APIs*/
uint32_t ssr;
uint32_t tr;
uint32_t dr;
do
{
ssr = RTC->SSR;
tr = RTC->TR;
dr = RTC->DR;
} while (ssr != RTC->SSR);

RTC_TimeStruct->SubSeconds = ssr;
RTC_TimeStruct->Hours = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_HOUR(tr) );
RTC_TimeStruct->Minutes = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_MINUTE(tr) );
RTC_TimeStruct->Seconds = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_SECOND(tr) );
RTC_DateStruct->WeekDay = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_WEEKDAY(dr) );
RTC_DateStruct->Date = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_DAY(dr) );
RTC_DateStruct->Month = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_MONTH(dr) );
RTC_DateStruct->Year = __LL_RTC_CONVERT_BCD2BIN( __LL_RTC_GET_YEAR(dr) );

#else

uint32_t first_read;
/* Get Time and Date*/
HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
/* make sure it is correct due to asynchronus nature of RTC*/
do {
first_read = RTC_TimeStruct->SubSeconds;
HAL_RTC_GetDate( &RtcHandle, RTC_DateStruct, RTC_FORMAT_BIN );
HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
} while (first_read != RTC_TimeStruct->SubSeconds);

#endif
/* calculate amount of elapsed days since 01/01/2000 /
calendarValue= DIVC( (DAYS_IN_YEAR3 + DAYS_IN_LEAP_YEAR)* RTC_DateStruct->Year , 4);
correction = ( (RTC_DateStruct->Year % 4) == 0 ) ? DAYS_IN_MONTH_CORRECTION_LEAP : DAYS_IN_MONTH_CORRECTION_NORM ;
calendarValue +=( DIVC( (RTC_DateStruct->Month-1)*(30+31) ,2 ) - (((correction>> ((RTC_DateStruct->Month-1)*2) )&0x3)));
calendarValue += (RTC_DateStruct->Date -1);
/* convert from days to seconds */
calendarValue *= SECONDS_IN_1DAY;
calendarValue += ( ( uint32_t )RTC_TimeStruct->Seconds +
( ( uint32_t )RTC_TimeStruct->Minutes * SECONDS_IN_1MINUTE ) +
( ( uint32_t )RTC_TimeStruct->Hours * SECONDS_IN_1HOUR ) ) ;
calendarValue = (calendarValue<<RTC_N_PREDIV_S) + ( RTC_PREDIV_S - RTC_TimeStruct->SubSeconds);
return( calendarValue );
}

jamesl-dm, 2019-05-17:
I think you need to put the SSR / TR / DR sampling loop in a critical section.

Oanerer, 2019-05-17:
@sidpat wrote:
> 
> 
> Will this be solved by
> 
> `HAL_RTCEx_DisableBypassShadow(&RtcHandle) // in rtc init function`
> 
> ```
> HAL_RTC_GetTime( &RtcHandle, time, RTC_FORMAT_BIN ); // Simple reading in RtcGetCalendarValue() function
> HAL_RTC_GetDate( &RtcHandle, date, RTC_FORMAT_BIN ); // no need for first_read check in while loop
> ```
> 
> ?
> 
> RTC shadow registers should ensure data integrity while reading.

In theory this could work, but using the shadow registers comes with some restrictions. From the reference manual, section 26.4.8: 

> When the BYPSHAD control bit is set in the RTC_CR register (bypass shadow
registers)
Reading the calendar registers gives the values from the calendar counters directly, thus
eliminating the need to wait for the RSF bit to be set. This is especially useful after exiting
from low-power modes (STOP or Standby), **since the shadow registers are not updated
during these modes**.

@jamesl-dm wrote:
> I think you need to put the SSR / TR / DR sampling loop in a critical section.

Adding a critical section does not solve anything. The RTC updates its registers autonomously, regardless of interrupts or whatever. We just happen to read the TR, DR and SSR registers in succession while in between the RTC is advancing one tick and updates its registers accordingly. 

To test a possible fix we have now replaced:

```
  /* Get Time and Date*/
  HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
  
   /* make sure it is correct due to asynchronus nature of RTC*/
  do {
    first_read = RTC_TimeStruct->SubSeconds;
    HAL_RTC_GetDate( &RtcHandle, RTC_DateStruct, RTC_FORMAT_BIN );
    HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
  } while (first_read != RTC_TimeStruct->SubSeconds);
```

with:

```
   /* make sure it is correct due to asynchronus nature of RTC*/
  do {
    first_read = RTC->SSR;
    HAL_RTC_GetDate( &RtcHandle, RTC_DateStruct, RTC_FORMAT_BIN );
    HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
  } while (first_read != RTC->SSR);
````

So far our HW_RTC_TestLoop() has not produced an error once while running for 45 minutes with this fix, whereas the original code produced an error within 30 seconds.

jamesl-dm, 2019-05-17:
Hypothetical example:

1. SSR is updating from 1024 to 1025. We read it on the clock edge, and get corrupt value 1026.
2. Read TR
3. Irq goes off wasting some time, RTC clock edge occurs, then we read DR
4. Read SSR and get correct value (1026)

Now TR and DR are out of sync.
Not sure how realistic this is though :)
Anyway, I use a critical section just in case.

Oanerer, 2019-05-17:
I understand what you're saying, but if we *also* have to worry about metastability of registers when reading from the APB busses in the MCU, we have far bigger issues than just reading a wrong date/time. 

The RTC registers will be synchronised to the APB bus clock, but we simply need several APB bus register reads before we have all the date/time information that together form a consistent 'calendar value'. The synchronisation mechanism in 'rtc-board.c' is intended to ensure just that.

sidpat, 2019-05-18:
@Oanerer wrote,

> In theory this could work, but using the shadow registers comes with some restrictions. From the reference manual, section 26.4.8:

The restriction is a small wait time to check if the RSF bit is set, which take 2 RTCCLK cycles i.e. 61us. Will this wait time be acceptable to remove a critical uncertainty? Just wondering what can possibly go wrong if we implement this using shadow registers.

jamesl-dm, 2019-05-20:
> I understand what you're saying, but if we also have to worry about metastability of registers when reading from the APB busses in the MCU, we have far bigger issues than just reading a 'calendar value'.

Section 26.3.8 of the datasheet suggests that metastability _is_ an issue:

>When the BYPSHAD bit is set to 1, the results of the different registers might not be
coherent with each other if an RTCCLK edge occurs between two read accesses to the
registers. Additionally, the value of one of the registers may be incorrect if an RTCCLK edge
occurs during the read operation. The software must read all the registers twice, and then
compare the results to confirm that the data is coherent and correct. Alternatively, the
software can just compare the two results of the least-significant calendar register.

>Just wondering what can possibly go wrong if we implement this using shadow registers.

The shadow registers don't update during sleep mode, so its a bit tricky to ensure that the reads actually return a value. Much faster to just do the synchronization manually.

YoannBo, 2019-05-20:
Hi,
Note that SSR wrap- around (e.g. from 1023 to 0) is the hardware source of TR increment. Likewise a wrap around on hour (from 23 to 0) is the hardware source for DR increment.
To if first read( value of SSR) is the same than second one, there can  not be metastability ssues cause TR/DR are stable from at least the first SSR read.

mluis1, 2019-05-20:
Thanks for the analysis and propositions.

If everyone agrees we will update the `rtc-board` files according to @YoannBo proposition.

Oanerer, 2019-05-20:
I'm not sure if mixing the normal HAL calls with LL API calls is the prettiest solution. Personally, I don't even have the LL drivers in my project as they are not used anywhere else in the stack nor my own code.

Obviously I'm biased, but I would simply use:
```
  /* make sure it is correct due to asynchronous nature of RTC*/
  do {
    first_read = RTC->SSR;
    HAL_RTC_GetDate( &RtcHandle, RTC_DateStruct, RTC_FORMAT_BIN );
    HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
  } while (first_read != RTC->SSR);
```

I've been successfully testing this for hours on end using the code to reproduce the issue I posted above and it's been running successfully in 14 nodes over the weekend using my normal code.

On another note: just found out this issue was already reported before: #560. Maybe reference that issue as well in your commit?

mluis1, 2019-05-23:
Thanks for the proposition @Oanerer.

I will update the different `rtc-board.c` files according to your proposition. This will avoid the usage of LL libraries.

closed 2019-06-07 by mluis1