Lora-net/LoRaMac-node #744: Not seeing Downlink

danielrsfreitas, 2019-05-17:
Hi all,

I am trying to connect to a LoRaWAN gateway and the gateway configuration is:
* AU915
* Uplink: 8 initial channels beginning in 915.2MHz, with 200KHz increments
* Downlink: 8 last channels beginning in  923.3MHz with 500KHz increments
* Rx1 = 5s e Rx2 = 6s

I am having issues to configure it following the specifications above. The uplink seems to work but the device is not seeing anything at the downlink channel but from gateway logs, it looks like it is coming down at  924.5MHz, 926.9MHz, 926.3MHz, 925.1MHz. Based on the Downlink channels I mentioned in bullets, does it make sense to receive data in this 4 channels, for example? How can I solve it ?

Logs below. Please note there is no downlink message. 

CLASS       : A

TX PORT   ###### ===== ClassA demo application v1.0.RC1 ==== ######

DevEui      : 30-30-32-34-52-35-6E-07

AppEui      : 70-B3-D5-7E-D0-01-94-52

AppKey      : 2B 7E 15 16 28 AE D2 A6 AB F7 15 88 09 CF 4F 3C

###### ===== JOINED ==== ######

ABP

DevAddr     : 70A0E027

NwkSKey     : 2B 7E 15 16 28 AE D2 A6 AB F7 15 88 09 CF 4F 3C

AppSKey     : 2B 7E 15 16 28 AE D2 A6 AB F7 15 88 09 CF 4F 3C

###### ===== MCPS-Request ==== ######

STATUS      : OK

###### ===== MCPS-Confirm ==== ######

STATUS      : OK

###### ===== UPLINK FRAME 1 ==== ######

CLASS       : A

TX PORT     : 2

TX DATA     : UNCONFIRMED

00 

DATA RATE   : DR_2

U/L FREQ    : 918200000

TX POWER    : 0

CHANNEL MASK: FFFF 

###### ===== MCPS-Request ==== ######

STATUS      : OK

###### ===== MCPS-Confirm ==== ######

STATUS      : OK

###### ===== UPLINK FRAME 2 ==== ######

CLASS       : A

TX PORT     : 2

TX DATA     : UNCONFIRMED

00 

DATA RATE   : DR_2

U/L FREQ    : 917800000

TX POWER    : 0

CHANNEL MASK: FFFF

danielrsfreitas, 2019-05-17:
Hi all,
I just noticed that my receive delay scheduling was wrong. The server requests for Rx1 5s and Rx2 6s. Modified here and downlink now works.

closed 2019-05-17 by danielrsfreitas

AhmadSarifudin, 2019-05-23:
Hi , I have same problem with downlink , are you connected to The thing network server or use the other server?

danielrsfreitas, 2019-05-23:
Hi AhmadSarifudin,

No, I am not using The thing network, I am using Everynet. In my case it was only a matter of changing the Receiving window delay. I guess the LoRaMAC-node is already compliant with TTN requirements. 

Would you share your ccmake configuration ?

lucaspeixotot, 2020-05-29:
Hi, Daniel, I'm having the same problem to receive downlink packets. I'm using Everynet as well and our config is the same. The only difference is that I changed my channel mask to 00FF 0000 0000 0000 FFFF because as my gateway is working from 915.2MHz to 916.6MHz some uplink packets are being sent out of this range. Downlink packets don't work both in class A and B. I configured my receive delay 1 and 2 calling LoRaMacMibSetRequestConfirm function with type MIB_RECEIVE_DELAY_1 and MIB_RECEIVE_DELAY_2. I think that I'm configuring what was supposed to configure. Do you have any idea how to solve it? My log is the same as yours.

danielrsfreitas, 2020-05-29:
Hi Lucas, it's been a while I don't work with LoRa so I can't recall very much. As I mentioned before, to solve my problem I only had to adjust the receiving window at the node. For example, it was 1s , 2s for the delay 1 and 2 respectively but Everynet uses 5s, 6s. Sorry that I can't give you the best answer and hope you overcome this hurdles.

lucaspeixotot, 2020-05-29:
Thanks, daniel. I created a new issue to try to solve my problem, thanks!