Lora-net/LoRaMac-node #745: RegionUS915.c does not support channel remasking

disk91, 2019-05-23:
The channel init create 72 channels for communication when the common gateway only have 8 listening channels. As a consequence most of the transmission are lost in US915 configuration.
We need to configure the node to only emit on the available channels on the gateways.
So we need after init to remap the channel really available like we do in EU868 as an example.

The solution I found to make it without impacting the whole file was to modify RegionUS915ChanMaskSet

I've added the following lines:
```C
       case CHANNELS_DEFAULT_MASK:
        {
            RegionCommonChanMaskCopy( NvmCtx.ChannelsDefaultMask, chanMaskSet->ChannelsMaskIn, CHANNELS_MASK_SIZE );
            break;
        }
       //  added lines >>>>>
        case CHANNELS_REINIT_MASK:
        {
            RegionCommonChanMaskCopy( NvmCtx.ChannelsMaskRemaining, chanMaskSet->ChannelsMaskIn, CHANNELS_MASK_SIZE );
            RegionCommonChanMaskCopy( NvmCtx.ChannelsDefaultMask, chanMaskSet->ChannelsMaskIn, CHANNELS_MASK_SIZE );
        	RegionCommonChanMaskCopy( NvmCtx.ChannelsMask, chanMaskSet->ChannelsMaskIn, CHANNELS_MASK_SIZE );
        	break;
        }
       /// <<<<<<
        default:
            return false;
```

and add this new type of ChanMask in Region.h file
```C
typedef enum eChannelsMask
{
    /*!
     * The channels mask.
     */
    CHANNELS_MASK,
    /*!
     * The channels default mask.
     */
    CHANNELS_DEFAULT_MASK,
	/*!
	 * The channels mask (added by disk91)
	 * Reinit all the mask
	 */
	CHANNELS_REINIT_MASK
}ChannelsMask_t;
```

By doing this the channel used for communication are restricted to the one given as parameter of the function.

mluis1, 2019-05-24:
Unfortunately your proposition is not LoRaWAN specification compliant. 
The specification mandates the usage of 64 channels for the US915/AU915 regions and even 96 channels for CN470 region.

As such the implementation provided here is compliant and moreover the other end-device stacks that I'm aware of are doing something similar as if they don't, they can't get LoRaWAN certification.

The usage of 8 channels gateway in regions like US915 are not defined by the LoRa-Alliance. The issue is that a lot of people do use those gateways in these regions mostly for cost reasons.

An end-device has no way to know if the gateways are listening on 8, 16, 64, 96 channels this means that it is up to the network server to instruct the end-device to use a sub-set of the available channels if necessary. The network server is the only one that has knowledge of the gateways capabilities connected to it.

This means that the end-devices must be able to handle all the channels available for a given region. Which means that when reaching a state where they detect a loss of connectivity they must re-enable all the available channels.
In such situations the user/network operator must accept that some frames are lost.

Please note that usually network servers send LinkAdrReq mac commands to the end-device as soon as it receives an uplink then the end-device will operate on expected channels.
Also most of the network servers send the CFList parameter of the JoinAccept in order to setup the channels mask.

Nowadays, this is the situation that we have. We can agree with it or not but, in case we disagree and we have a better solution to solve the issue then everyone is welcome to propose their solution to the LoRa-Alliance technical committee that will discuss it and if quorum is achieved then integrate it in future specifications revisions.

disk91, 2019-05-24:
Thank you for sharing this, this is really relevant and I agree on what you tell. I also think LoRa-Alliance should really review this part of the spec ... even if I understand the legal context fro using the US915 Band I assume we can do better.

What I do not clearly understand is why we could change a setting after a join and not putting it staticly. That make no sense and I assume to be compliant you need to use 64 channels for joinging and once join. So the CFList may be a 64 channels map on the 72 to respect the regulation. Isn't it ?

Meaning in US915 only gateways with 64 channels could be relevant. Quite interesting in term of network deployment costs...

So that said, I'll turn my question differently: If I proposed the change above it is to limit the modifications in the US915 file until we have something better. Because I was a bit surprized by the following functions:
```C
uint8_t RegionUS915NewChannelReq( NewChannelReqParams_t* newChannelReq )
{
    // Datarate and frequency KO
    return 0;
}

int8_t RegionUS915TxParamSetupReq( TxParamSetupReqParams_t* txParamSetupReq )
{
    return -1;
}

uint8_t RegionUS915DlChannelReq( DlChannelReqParams_t* dlChannelReq )
{
    return 0;
}
LoRaMacStatus_t RegionUS915ChannelAdd( ChannelAddParams_t* channelAdd )
{
    return LORAMAC_STATUS_PARAMETER_INVALID;
}

bool RegionUS915ChannelsRemove( ChannelRemoveParams_t* channelRemove  )
{
    return LORAMAC_STATUS_PARAMETER_INVALID;
}
```

Can someone tells me why we have no implementation for these functions ?

So reading your comment, I assume this function is called on LinkAdrReq (make sense for sure) and will correctly set the channels. This is by-the-way more to inform reader of this discussion than a question as the answer in included.

uint8_t RegionUS915LinkAdrReq( LinkAdrReqParams_t* linkAdrReq, int8_t* drOut, int8_t* txPowOut, uint8_t* nbRepOut, uint8_t* nbBytesParsed )

Now, I need to see if TTN is supporting this feature basically related to each of the gateways making it a bit complex to manage in a network server point of view with an heterogeneous type of gateways.

mluis1, 2019-06-13:
The LoRaWAN specifications says that for fixed channel plans the referenced functions must not be handled.

Example:
[image]

These functions are implemented empty and returning an error code on those regions as we wanted to keep as much as possible the same file layout for all regions.

closed 2019-06-13 by mluis1