Lora-net/LoRaMac-node #746: Deadlock at huge data payload retransmissions

miran-rajsek, 2019-05-24:
Hi, 

I've come across a deadlock in OnMacStateCheckTimerEvent that blocks sending further uplinks. My lorawan stack is v4.4.1, region EU868.

It happens at huge data payload, class A confirmed uplink retransmissions.
As far as I can see, data rate for next retransmission is decreased and consequently payload length validation fails.

It occurs in spite of that I'm calling LoraMacQueryTxPossible at the beginning and it returns LORAMAC_STATUS_OK. 

The deadlock starts with
OnTxDelayedTimerEvent() -> ScheduleTx( true ) -> ValidatePayloadLength returning false.
OnMacStateCheckTimerEvent then loops in LoRaMacState LORAMAC_TX_RUNNING, looks like doing nothing but starting MacStateCheckTimer (https://github.com/Lora-net/LoRaMac-node/blob/v4.4.1/src/mac/LoRaMac.c#L1422).

My case description:
- Confirmed uplink with data payload size 51, FCnt=0, sent on SF=12, good RSSI and SNR.
- NMS sends MAC downlink MAC.LinkADRReq.DataRate_TXPower.DataRate=5, i.e. NMS wants my device to send next uplink on SF7.
- At this point I'm deleting my device from NMS, but still monitoring device operation in the console. Absence of confirmation downlinks is going to cause retransmissions.   
- FCnt 1 uplink ends up with LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT as expected.
- Upon sending FCnt 2, the deadlock occurs at the second retransmission.

According to device console output it seems to me that lorawan library was decreasing data rate between retransmissions but didn't re-validate payload length before deciding for the lower data rate.  If retransmissions are about sending the same content, decreasing data rate between retransmissions should stop at the level determined by the content size. 

Currently my application tackles data payload size control by calling LoraMacQueryTxPossible, setting zero data payload and unconfirmed uplink in negative case, thus giving mac level communication higher priority over application data. 

Additionally I'm thinking about hard-coding a reasonable data payload size limitation, focusing on EU868 region only. I would be OK with, for example, 30 bytes, thus leaving approx. 20 bytes for mac level. Was looking around, but couldn't find any fixed number for maximum MAC level payload size. Do we have any rules of thumb, best practices, etc?

Thanks, 
Miran

mluis1, 2019-05-24:
Thanks for the report and analysis.

You are right the re-transmission  procedure doesn't take in account the datarate decrease.

In the upcoming LoRaWAN specifications (1.0.4 and 1.1.x) the datarate decreasing possibility has been completely removed. You may already check this by looking at the released LoRaWAN 1.1.0 specification.

This problem can also be seen on current develop and will be present as well on the upcoming 4.4.2 release.

All this means that we have only 2 choices to solve the problem.

1. Reduce the maximum payload size to a value that can be used at the lowest datarate.
According to the regional parameters specification the maximum allowed payload for SF12 on EU868 region is 51 bytes if no `FOpts` field is present. The `FOpts` field maximum size is 15 bytes. So, the worst case is 51 - 15 = 36 bytes. Then **36** bytes is the maximum payload size that one can use in order to take in account the datarate decreasing.
2. Change the number of retries to a maximum of **1**. (Please note that this is the approach used for the future LoRaWAN specifications.)
By doing this the MAC layer will only transmit once the frame. In case the `ACK` isn't received then the application can decide to transmit the same application payload.
One of the biggest advantages of this aproach is that every single frame will get its frame counter incremented which improves the protocol security.
*With this option the payload size is not anymore an issue.*

miran-rajsek, 2019-05-27:
Thanks, that's very useful info.

closed 2019-05-27 by miran-rajsek