Lora-net/LoRaMac-node #748: OnTxDelayedTimerEvent continuosly called for Class-A unconfirmed

adibacco, 2019-06-03:
I have a node running STM32 LRWAN 1.2.1. The node is Class A and it  is configured with DUTY CYCLE OFF. I'm sending UNCONFIRMED messages that are always correctly received but after thousands of messages sent , the node starts to duplicate every message and I saw that **OnTxDelayedTimerEvent** is being called all the times. What is the purpose of **OnTxDelayedTimerEvent** ?

fahad80, 2019-06-05:
What region you are using?

**OnTxDelayedTimerEvent** is called during CONFIRMED uplinks if it fails to receive ACK or if you get LORAMAC_STATUS_DUTYCYCLE_RESTRICTED.

adibacco, 2019-06-05:
I’m using region EU868 but I never send CONFIRMED messages and I set
DUTY_CYCLE to false in lora.c then I don’t understand why this OnTxDelayed
is being called. Anyway, just for demo, I’m transmitting every 5 secs.

adibacco, 2019-06-06:
Today I discovered that the messages duplication and **OnTxDelayedTimerEvent**  continuos calls are triggered by an RxError messages coming out in OnExError function.

mluis1, 2019-06-13:
Can we close this issue?

adibacco, 2019-06-13:
Yes, sorry

closed 2019-06-13 by mluis1