Lora-net/LoRaMac-node #749: EN 300 220 testing

KlemenBlokar, 2019-06-11:
Hi, I need help regarding using the LoRaMac-node for certification of devices under EN 300 220. I received a list of requirements for certification testing:
```
3. Samples documentation
- Short document (single A4 page) with instructions on how to operate samples
- Software for controlling the samples and necessary instructions

4. Configuration of the samples
On all samples manual configuration of operating modes has to be enabled. Following modes have to be available:
- Selection of transmission channel
- Continuous transmission with modulation
- Continuous transmission without modulation
- Adjustment of transmission power
- Receive mode only
- Transmission of known number of messages and monitoring of how many messages was received (to see how many packages was received)
- Transmission in burst mode (0.5 s ON, 0.5 s OFF)
- For testing purposes, the duty cycle restrictions are excluded, so the EUT can transmit constantly. 
```

I know that most of the requirements can be met by entering test mode and issuing appropriate MAC commands. But I do not know, are burst mode, continuous rx and tx modes supported too? And I have no idea how to provide useful documentation on how to enter those modes. Has anyone dealt with this before? How did you provide those modes of operation?

Any help would be very appreciated.

mluis1, 2019-06-13:
I would recommend that you post your question on the [Semtech LoRa Developers portal](https://lora-developers.semtech.com/knowledge-base/forum/) forum. You will get better support than here concerning this subject.

closed 2019-06-13 by mluis1

fahad80, 2019-06-14:
You can do that directly using Radio functions. At least thats how I did it.

KlemenBlokar, 2019-06-17:
Ok, so you just prepared a special firmware that bypassed LoRaMac-node library for those tests?